don/cordova-plugin-ble-central #188: iOS write command stopped working with Cordova 6.0.0

Janckk, 2016-02-23:
Hi,

so everything was working fine until the newest update of cordova and xCode on iOS. Now I'm experiencing no errors but "write" fails silently as well I get no "notifications" from startNotification. Connects, scan and everything else works fine.

Did anyone experience this as well?

don, 2016-02-25:
It sounds like Cordova 6 iOS broke something. Maybe you can drop back to an older version for a bit?

I'll be working on a new BLE project next week, so a lot of these issues should get fixed soon.

zabojad, 2016-02-25:
If we can't have Cordova 5 and Cordova 6 supported together, maybe we could have one branch of this repo for each?

don, 2016-02-25:
@zabojad the plan is to support both. I just need to block off some time to duplicate and fix issues. Are you seeming this problem too?

zabojad, 2016-02-25:
@don Not yet but I shall face it soon as I plan to upgrade few apps to Cordova 6.0...

don, 2016-02-27:
@Janckk I can't duplicate this. I've been working on an app today with many writes and they're all working.

I'm running 
- cordova 6.0.0 
- cordova platform ios 4.0.1
- cordova-plugin-ble-central 1.0.4
- OS X 10.10.5
- Xcode Version 7.2.1 (7C1002) 

Can you ensure that the BLE plugin and cordova ios platform are up to date?
Can you provide a simple example that duplicates the problem?

don, 2016-03-19:
Can't duplicate.

@Janckk if you're still seeing this, try updating your native project. If you have more info, post here and I'll reopen the issue.

closed 2016-03-19 by don

snowbreeze, 2016-03-28:
I am getting Writing is not permitted.