don/cordova-plugin-ble-central #189: Android: startScan, and startNotification callback ArrayBuffer not convertible to string

wanjungao, 2016-02-25:
I print out arraybuffer 

advertising data {"0":2,"1":1,"2":6,"3":3,"4":3,"5":1,"6":174,"7":18,"8":8,"9":65,"10":101,"11":114,"12":111,"13":45,"14":66,"15":56,"16":50,"17":55,"18":69,"19":66,"20":66,"21":56,"22":50,"23":56,"24":51,"25":48,"26":0,"27":0,"28":0,"29":0,"30":0,"31":0,"32":0,"33":0,"34":0,"35":0,"36":0,"37":0,"38":0,"39":0,"40":0,"41":0,"42":0,"43":0,"44":0,"45":0,"46":0,"47":0,"48":0,"49":0,"50":0,"51":0,"52":0,"53":0,"54":0,"55":0,"56":0,"57":0,"58":0,"59":0,"60":0,"61":0}

notice that the advertising data has lots of 0s. Is it intended behavior?

startNotification call back
{"0":123,"1":34,"2":115,"3":116,"4":97,"5":116,"6":117,"7":115,"8":34,"9":58,"10":50,"11":48,"12":48,"13":44,"14":34,"15":98,"16":111,"17":100,"18":121,"19":34}

after i translate to string value is {"status":200,"body". So it appears the data from notification is cut off.

In both cases, ios version works well. IOS callbacks use json string as parameter. 

Also, is it a reason to have different data signature between ios and android?

don, 2016-02-25:
That looks normal for an advertising info. Everything is sent in the ArrayBuffer.

The success callback for startNotification should be called with an ArrayBuffer on iOS and Android. If you're seeing different behavior, iOS might be broken with the latest update. (Could be related to #188?)

As far as the data being cut off, that depends on the hardware. Generally iOS handles large data better. The characteristic size is 20 bytes. Some hardware will send blob data up to 500 bytes, but it's hardware and software dependent and handled at a low level is iOS and Android.

You data is 20 bytes, so I think that's the most you'll get on that Android device.

wanjungao, 2016-02-25:
I see. I started with iOS, and thought the normal data returned was JSON, and coded accordingly. 

Thanks. And good to know about the notification data size issue is related to hardware.

closed 2016-02-25 by wanjungao

don, 2016-02-25:
What version of Cordova are you using where your seeing JSON returned in the callback?

wanjungao, 2016-02-25:
6.0.0, but I got my plugin several days ago. So it could have been fixed since then.