don/cordova-plugin-ble-central #190: Error: "Could not find service with UUID..."

leo3linbeck, 2016-02-26:
First, thanks for the great library! It has been terrific to have it available - I know how much work it is to do something like this, so kudos to you for making it available open source.

I'm able to scan and connect with a BLE device. However, when I try to write, I get the following error:

`2016-02-25 20:45:55.236 Brevitest[3204:1916506] Could not find service with UUID 5998A6F2-2377-4919-BED5-B1AF98C28985 on peripheral with UUID 884FA2EE-4A39-1761-9384-ECE0F336463A
`
I know both the UUIDs are correct, as here is the callback result from the scan call:

`2016-02-25 20:44:10.526 Brevitest[3204:1916506] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = Kutuzov;
        kCBAdvDataServiceUUIDs =         (
            "5998A6F2-2377-4919-BED5-B1AF98C28985"
        );
    };
    id = "884FA2EE-4A39-1761-9384-ECE0F336463A";
    name = Kutuzov;
    rssi = "-56";
}
`
As you can see, the UUIDs match. Here's the code that calls the write function:

`  var service_uuid =        '5998A6F2-2377-4919-BED5-B1AF98C28985';
  var characteristic_uuid = '75395B63-67CB-4684-BC2B-52985A28B249';

  $scope.sendTestMessage = function(indx) {
      var device = $scope.devices[indx];

      if (device.connected) {
          $ionicPlatform.ready(function() {
              $cordovaBLE.write(device.bluetooth_id, service_uuid, characteristic_uuid, stringToBytes('Test'))
                .then(function(result) {
                  device.connected = false;
                })
                .catch(function(error) {
                  ngNotify.set('Error sending message' + JSON.stringify(error), 'error');
                });
          });
      }
 };
`
I've tried it both with and without the $ionicPlatform.ready wrapper, and I've confirmed that all of the arguments are as they should be. I'm stumped as to where to go from here, or even what to look for. Any help would be appreciated.

(NB: this happens on both 5.4.1 and 6.0.0 of Cordova)

don, 2016-02-26:
@leo3linbeck When you connect to the device, you should get a JSON object back that describes the peripheral in the success callback. Can you print that object out and paste the results here?

    JSON.stringify(peripheral, null, 2);

leo3linbeck, 2016-02-26:
@don
As you wish:

`{
  "characteristics": [
    {
      "value": {},
      "properties": [
        "Read"
      ],
      "characteristic": "2A25",
      "isNotifying": false,
      "service": "180A"
    },
    {
      "value": {},
      "properties": [
        "Read"
      ],
      "characteristic": "2A27",
      "isNotifying": false,
      "service": "180A"
    },
    {
      "value": {},
      "properties": [
        "Read"
      ],
      "characteristic": "2A26",
      "isNotifying": false,
      "service": "180A"
    },
    {
      "value": {},
      "properties": [
        "Read"
      ],
      "characteristic": "2A28",
      "isNotifying": false,
      "service": "180A"
    },
    {
      "value": {},
      "properties": [
        "Read"
      ],
      "characteristic": "2A29",
      "isNotifying": false,
      "service": "180A"
    },
    {
      "value": {},
      "properties": [
        "Read"
      ],
      "characteristic": "2A24",
      "isNotifying": false,
      "service": "180A"
    }
  ],
  "id": "884FA2EE-4A39-1761-9384-ECE0F336463A",
  "rssi": -49,
  "advertising": {
    "kCBAdvDataIsConnectable": true,
    "kCBAdvDataServiceUUIDs": [
      "5998A6F2-2377-4919-BED5-B1AF98C28985"
    ],
    "kCBAdvDataLocalName": "Kutuzov"
  },
  "name": "Kutuzov",
  "services": [
    "180A"
  ]
}`

Cheers,
L3

don, 2016-02-26:
I see service `5998A6F2-2377-4919-BED5-B1AF98C28985` being advertised, but it is not listed in the discovered services. That's why you're seeing the error.

Can you see the service using other tools or devices?

leo3linbeck, 2016-02-26:
Hmmm. No, I can't. Well, that's good news on two fronts:
1. It's not a problem with the BLE plugin (of course, you already knew that) ;-)
2. It gives me a place to focus attention.

I guess the question is: why wouldn't the BLE chip list that service? If you have any ideas of where to look, it would be much appreciated. But since it doesn't appear to impact your code, there's no expectation you will.

Thanks, and cheers,
L3

leo3linbeck, 2016-02-26:
Well, I figured it out ;-).
The problem was that the iPhone was caching the old service/characteristic data and not updating it. I turned off Bluetooth, then turned it back on, and the service and characteristic both magically appeared.
Appreciate the help. Cheers,
L3

don, 2016-02-26:
Do you control the peripheral?

don, 2016-02-26:
Do you control the peripheral?

Service Changed 0x2a05 characteristic is not present in the Generic Attribute Service 0x1801iOS will cache all attributes even if the peripheral changes them. Implementing Service Changed may fix this.

closed 2016-02-27 by don

LoFiRobot, 2017-02-06:
Hello Don and leo3linbeck,

though the issue is marked as closed i have similar problem and cant`t find my way around it.
My app works correctly on Android but on iOS I am recieving error "Could not find service with UUID...."

I am using two different BLE modules (HM-10 and DFRobot Bluno) both giving the same error.
Turning bluetooth on/off on the device doesnt help.
Do you have any idea how to work this out?

Scanning JSON output below.

[Log] {"id":"609D04F3-6A9E-42D8-9727-AC344A2487BE",
"rssi":-51,
"advertising":{"kCBAdvDataIsConnectable":true,
"kCBAdvDataLocalName":"HM Soft",
"kCBAdvDataServiceUUIDs":["FFE0"],
"kCBAdvDataServiceData":{"B000":{}}
,"kCBAdvDataTxPowerLevel":0,
"kCBAdvDataManufacturerData":{}},
"name":"LOFI_Robot5"} (index.js, line 63)

[Log] {"id":"2AF01B50-F902-4D6F-ABE7-B4C24EC08E24",
"rssi":-61,
"advertising":{"kCBAdvDataLocalName":
"Bluno","kCBAdvDataServiceUUIDs":["DFB0","1812"],
"kCBAdvDataTxPowerLevel":0,
"kCBAdvDataIsConnectable":true},"name":"Bluno"} (index.js, line 63)

LoFiRobot, 2017-02-06:
OK i got it working :-)

In my case the solution was to change the full UUIDs to shrot versions.
So for HM-10 module:

instead of:
  serviceUUID: '0000ffe0-0000-1000-8000-00805f9b34fb',
  txCharacteristic: '0000ffe1-0000-1000-8000-00805f9b34fb', 
  rxCharacteristic: '0000ffe1-0000-1000-8000-00805f9b34fb'

I use:
  serviceUUID: 'FFE0',
  txCharacteristic: 'FFE1', 
  rxCharacteristic: 'FFE1'

jaean123, 2017-07-13:
I had same issue, but it works now.

Thanks @LoFiRobot @leo3linbeck

jaean123, 2017-07-13:
I had same issue. It works now.

Thanks @don @LoFiRobot @leo3linbeck