don/cordova-plugin-ble-central #191: Andriod 6- plugin not working

MuraliKrishnaDev, 2016-02-29:
Hi,

Please guide me how run this plugin in Marshallow device,

Thanks and regards,
Murali

don, 2016-02-29:
Support for Android 6 will be in the next release. Until then, edit `config.xml` and target sdk 22.

    <preference name="android-targetSdkVersion" value="22"/>

don, 2016-02-29:
Duplicates of #182

MuraliKrishnaDev, 2016-03-01:
Hi,
After changing the config.xml,I am unable to run the evothings.ble.startscan function and I am not getting the results as in API22,
I am scanning beacons using marshallow device,
Please help me regarding this,
Best Regards,Murali

Date: Mon, 29 Feb 2016 07:43:27 -0800

don, 2016-03-01:
You want ble.scan or ble.startScan. Evothings is a different plugin. See the README for more API details.

MuraliKrishnaDev, 2016-03-01:
Hi,
Thanks for the inputs,
I tried both ble.scan and ble.start scan in Visual studio 2015 and cordova 5.4.0 for MotoG2 device Marshallow updated(Andriod 6),
The following is the codeIndex.js-----------
    document.addEventListener( 'deviceready', onDeviceReady.bind( this ), false );

    function onDeviceReady() {

        // Handle the Cordova pause and resume events

        document.addEventListener( 'pause', onPause.bind( this ), false );

        document.addEventListener( 'resume', onResume.bind( this ), false );

        

        document.getElementById('click').addEventListener('click', BluetoothConnection, false);

        

        // TODO: Cordova has been loaded. Perform any initialization that requires Cordova here.

    };
function BluetoothConnection() {

        try {
 ble.scan([], 1000, function (device) {

                alert(JSON.stringify(device));

            },

            function fail() { alert('fail'); });

}

 ble.startScan(

             function (device) {

                 

                 alert(JSON.stringify(device));

             },

             function fail() {

                 // Report error. 

                 alert('fail');

             }

             );

        } catch (e) {

            debugger;

            throw e;

        }

I failed to get the data from the above code,Requesting you to help me in this regard(My plan is fetch beacon data from ble.startscan),
Thanks in anticipation,
Best Regards,Murali
Date: Mon, 29 Feb 2016 20:55:18 -0800

choijeonghwan, 2016-03-18:
config.xml 
< preference name="android-targetSdkVersion" value="22" />

AndroidManifest.xml
< uses-sdk android:minSdkVersion="14" android:targetSdkVersion="22" />

galaxy note5 marshmallow 6.0.1 working ok

jas67, 2016-03-18:
Just setting the targetSdkVersion to 22 didn't do it for me.

I also needed to enable location services permission for the plugin by adding the following to plugin.xml:

<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
<uses-permission android:name="android.permission.ACCESS_LOCATION_EXTRA_COMMANDS" />

don, 2016-04-15:
Fixed in version 1.0.5

closed 2016-04-15 by don