don/cordova-plugin-ble-central #196: ble.stopNotification() on success  return issue

cajay01, 2016-03-11:
In ble I used megster plugin in that ble.stopNotification() on success return it returns me an array buffer
but currently I updated the plugin to ble-central-plugin and for same call now it returns me "OK"
so is this plugin change?because in that array buffer I get some values that I am not getting now since it returns only OK .
What should I do to fix this issue?

don, 2016-03-19:
It sounds like this is working as intended. [ble.stopNotification](https://github.com/don/cordova-plugin-ble-central#stopnotification) should return nothing (or perhaps OK) in the success callback to indicate that the notification has been removed. 

What data were you expecting in stopNotification?

don, 2016-04-15:
[ble.stopNotification](https://github.com/don/cordova-plugin-ble-central#stopnotification) works as intended

Ignore any data passed to the success callback. Android will pass "OK", iOS will pass null. (This is the default Cordova behavior.)

https://github.com/don/ble-issue196

closed 2016-04-15 by don