don/cordova-plugin-ble-central #198: galaxy s6 android marshmallow 6.0.1, galaxy note5 marshmallow 6.0.1 not work

choijeonghwan, 2016-03-15:
android update 6.0.1 o/s after cordova-plugin-ble-central not working.

android 5.1.1 ==> working ok
android 6.0.1 ==> not working

please.  ASAP problem solution.

jplourenco, 2016-03-15:
Android 6 requires the location services to be enabled, in order to detect bluetooth devices.

choijeonghwan, 2016-03-15:
android 6 only bluetooth enable.
It is possible bluetooth detect.

source code 
ble.scan([], 5, app.onDiscoverDevice, app.onError);
not work

jplourenco, 2016-03-15:
Edited my previous answer

choijeonghwan, 2016-03-15:
Referring to article tried , but does not work

choijeonghwan, 2016-03-18:
config.xml
<preference name="android-targetSdkVersion" value="22"/>  

AndroidManifest.xml
 <uses-sdk android:minSdkVersion="14" android:targetSdkVersion="22" />

Properties for MainActivity -> Android -> Project build Target
Android 5.1.1 select

galaxy note5 marshmallow 6.0.1 working ok

choijeonghwan, 2016-03-18:
Properties for MainActivity --> Android --> Android 5.1.1 select

config.xml 
< preference name="android-targetSdkVersion" value="22" />  

AndroidManifest.xml
< uses-sdk android:minSdkVersion="14" android:targetSdkVersion="22" />

galaxy note5 marshmallow 6.0.1 working ok

closed 2016-03-18 by choijeonghwan