don/cordova-plugin-ble-central #201: Peripheral from connect doesn't have enough information

scottmahr, 2016-03-18:
Scanning works well, then I connect to a device (and I can tell it does actually connect). The issue is that the peripheral that I get back only has {"deviceHandle":7,"state":2}, and not all the information about the services and characteristics. 

I also try to do a read, and get an error back saying "invalid action".

Am I doing something wrong?

Thanks,

Scott

don, 2016-03-19:
That looks like something isn't being handled correctly. 

Can you run `adb logcat` to see if there is a stack traces when you are connecting?

Can you try connecting to that peripheral with nRF Master Control Panel? https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en

scottmahr, 2016-03-19:
What should I filter for using adb logcat?

don, 2016-03-19:
In nRF Master Control panel do you see services and characteristics listed after connecting?

For `adb logcat` I usually run without filters and just look for Java stack traces.  If you find something interesting, post here or email me.

scottmahr, 2016-03-19:
I found some stuff that hopefully makes sense to you:

E/bt-btif ( 4551): property type:241, len:0 is invalid
E/        ( 4551): ### ASSERT : external/bluetooth/bluedroid/main/../btif/src/btif_dm.c line 4718 failed to save remote device manufacturer (1) ###
E/BluetoothRemoteDevices( 4551): devicePropertyChangedCallback: bdDevice: EB:1D:8E:44:D0:D1, value is empty for type: 241
E/CordovaWebViewImpl( 7446): CordovaWebView: TIMEOUT ERROR!
E/bt-btif ( 4551): property type:241, len:0 is invalid
E/        ( 4551): ### ASSERT : external/bluetooth/bluedroid/main/../btif/src/btif_dm.c line 4718 failed to save remote device manufacturer (1) ###
E/BluetoothRemoteDevices( 4551): devicePropertyChangedCallback: bdDevice: EB:1D:8E:44:D0:D1, value is empty for type: 241
E/bt-btm  ( 4551): add new WL entry in bg_dev_list change : 0  own type: 0
E/bt-btm  ( 4551): btm_update_dev_to_white_list
E/bt-btm  ( 4551): btm_ble_start_auto_conn start : 1, 0
E/bt-btm  ( 4551): btm_execute_wl_dev_operation
E/bt-btm  ( 4551): btm_ble_count_unconn_dev_in_whitelist count:1
E/bt-btm  ( 4551): btm_ble_start_auto_conn start : 1, 0
E/bt-btm  ( 4551): btm_execute_wl_dev_operation
E/bt-btm  ( 4551): btm_ble_count_unconn_dev_in_whitelist count:0
E/bt-btif ( 4551): remote name callback registration fail.
E/BluetoothRemoteDevices( 4551): aclStateChangeCallback: State:Connected to Device:EB:1D:8E:44:D0:XX
E/bt-btif ( 4551): No More Service found

EB:1D:8E:44:D0:D1 is the device I am connecting to. 

Thanks, 

Scott

scottmahr, 2016-03-19:
And yes, I can see all the services and characteristics in MCF.

don, 2016-03-19:
I was hoping for a Java stacktrace. Those errors are in native code. I'm not sure I can do anything about that.

Are you calling ble.scan before calling ble.connect?
Does your Cordova code with other BLE devices on this phone?
Does your Cordova code work on different phones with this BLE device?
Does restarting Bluetooth (or rebooting the phone) make a difference?

Trying to figure out if it's a device specific or phone specific issue.

don, 2016-04-15:
I think this is a device specific issue. If you get a Java stacktrace from `adb logcat` attach it here and I'll reopen this.

closed 2016-04-15 by don