don/cordova-plugin-ble-central #203: Problems with device advertising data

zabojad, 2016-03-21:
Hi there !

I'm trying to use 

On iOS, the scan method success callback is called several times for the same Emerald device and I can't see the services (there should be two services) described in the advertising data:
```
2016-03-21 15:30:55.767 TestEmerald[398:238695] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "T0JU4AEADwBNAQAFogMBAAAAAAAAAAAAf/8=";
        };
    };
    id = "458E3376-60D6-0C9B-B124-4EA36DB29AD0";
    rssi = "-69";
}
2016-03-21 15:30:57.061 TestEmerald[398:238695] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = "EMERALD  010-F004D";
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "T0JU4AEADwBNAQAFogMBAAAAAAAAAAAAf/8=";
        };
    };
    id = "458E3376-60D6-0C9B-B124-4EA36DB29AD0";
    name = "EMERALD  010-F004D";
    rssi = "-71";
}
2016-03-21 15:31:17.808 TestEmerald[398:238695] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = "EMERALD  010-F004D";
    };
    id = "458E3376-60D6-0C9B-B124-4EA36DB29AD0";
    name = "EMERALD  010-F004D";
    rssi = "-76";
}
2016-03-21 15:31:17.812 TestEmerald[398:238695] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
    };
    id = "458E3376-60D6-0C9B-B124-4EA36DB29AD0";
    name = "EMERALD  010-F004D";
    rssi = "-76";
}
2016-03-21 15:31:20.403 TestEmerald[398:238695] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "T0JU4AEADwBNAQAFogMBAAAAAAAAAAAAf/8=";
        };
    };
    id = "458E3376-60D6-0C9B-B124-4EA36DB29AD0";
    name = "EMERALD  010-F004D";
    rssi = "-67";
}
2016-03-21 15:31:24.295 TestEmerald[398:238695] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = "EMERALD  010-F004D";
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "T0JU4AEADwBNAQAFogMBAAAAAAAAAAAAf/8=";
        };
    };
    id = "458E3376-60D6-0C9B-B124-4EA36DB29AD0";
    name = "EMERALD  010-F004D";
    rssi = "-72";
}
2016-03-21 15:31:54.556 TestEmerald[398:238695] stopScanTimer
```

Why do I get several times the scan success callback called with different rssi values?

Why are the services not appearing in the advertising data?

Note that on Android, I do not get the scan success callback called more than once...

zabojad, 2016-03-25:
I got this problem fixed.

closed 2016-03-25 by zabojad

don, 2016-03-25:
I've seen this on iOS where I get multiple scan results as the rssi changes. Occasionally I think I get multiple results for scan and scan response packets. I deduce these in my UI using the uuid. (There might be some iOS scan options to make this better, but they're not exposed in the plugin yet)

I'm relying on Android and iOS to do the scanning so you're seeing the OS behavior. 

The Emerald might have 2 services, but it may not be advertising them. You should see all the services when you connect to the peripheral. You can use a program like LightBlue or nRF Master Control panel to view the advertising data.

don, 2016-03-25:
@zabojad how'd you fix this?

zabojad, 2016-03-25:
@don I've fixed half of the problems I've described actually... I indeed get the advertising data on connecting to the Emerald.

I have not fixed the iOS issue which is beyond my vodoo ^^... Perhaps we should let this ticket open until we find some fix...

If that can help, I'm in direct contact with the Emerald engineering team so if you tell me what exact questions I should ask them for you to have the right pieces of information about what scan options we should use to avoid those multiple results, then we might have some happy ending on this problem too ^^...