don/cordova-plugin-ble-central #204: Ability to disable bluetooth

RangerMauve, 2016-03-22:
I seem to be running into issues where bluetooth stops working properly after a long interval of time (scanning no longer shows results), and I'd like to be able to restart bluetooth by toggling it off and on again. Would it be possible to add the ability to turn it off to match the ability to turning it on with `enable`?

don, 2016-03-25:
There's not a good way to programmatically disable and enable Bluetooth. Both iOS and Android want user interaction during the process. I can't recall the details but I couldn't find a good way to add a disable function without adding a UI. 

If you want to do this yourself on Android see http://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#disable() and http://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#enable()

RangerMauve, 2016-03-28:
Thanks for the response. Would I be correct to assume that an HTML button wouldn't be sufficient?

RangerMauve, 2016-03-30:
I looked into it, and it seems that the UI is just a strong suggestion from Google, it isn't required by the code. All you'd need is the `BLUETOOTH_ADMIN` permission and it'll work. This means that the UI requirement can be pushed on the users of the library. 

i.e. 

```html
<button id="mybutton">Disable Blueooth</button>
```

```javascript
mybutton.onclick = function(){ ble.disable(etc, etc) };
```

It seems that [other plugins](https://github.com/randdusing/cordova-plugin-bluetoothle#disable) have the functionality with the same assumption.

Would a PR for this feature be welcome to speed implementation along?

RangerMauve, 2016-03-30:
Additionally, I'd need a way to enable bluetooth without that prompt, would it be OK to just use the `enable()` method rather than showing the prompt on Android?

closed 2016-04-13 by RangerMauve