don/cordova-plugin-ble-central #205: Observer.

CharCorDer, 2016-03-23:
Is it possible to use my app as receiver, so i can get data from device which is broadcasting?

I watched all examples and i see only, that i can get service data with connection/pairing.
But if that device is already paired with some other device and switched to broadcasting mode, so other devices can get data?

don, 2016-03-25:
You get the broadcast data while scanning. 

The plugin can connect to multiple devices. You can scan while you are connected to devices. 

This example gets the temperature from the advertising data while scanning. https://github.com/don/ITP-BluetoothLE/tree/master/phonegap/thermometer_v2

It works with this Arduino code that broadcasts the temperature characteristic. https://github.com/don/ITP-BluetoothLE/blob/master/arduino/Thermometer_v2/Thermometer_v2.ino

CharCorDer, 2016-03-29:
But is it possible to read characteristic of service without connecting to that device? i want to read data from polar heartbeat which is already connected to other device watch/phone etc

don, 2016-03-29:
You need to connect to the peripheral to read a characteristic's value.

CharCorDer, 2016-03-29:
I use Polar Beat app https://play.google.com/store/apps/details?id=fi.polar.beat, and i pair sensor with tablet, then i switch option Visibility in that app. And then i want to read heartbeats with my phone, but it's not possible to connect to heart rate sensor in that moment.

don, 2016-03-30:
I don't think you can do what you want. Bluetooth 4.1 does allow a peripheral to be connected to multiple central devices, but your hardware would need to explicitly support this.

closed 2016-03-30 by don