don/cordova-plugin-ble-central #206: cant find variable ble

nikitavarat, 2016-03-25:
Hello there,
I am using IntelXDK for developing hybrid app. I am trying to run sample BLE app given by IntelXDK which is using this plugin. On Android phone it works perfectly but on iPhone with OS 9.2.1 it is raising exception 'can't find variable ble'.
I am using same program for Android & iPhone.

don, 2016-03-25:
'Can't find variable ble' usually means the plugin isn't installed, there's a JavaScript error or ble is being called before deviceready is fired. 

This might be an IntelXDK specific problem. 

Can you try removing and reinstalling the plugin?

nikitavarat, 2016-03-30:
Hello,
Reply I got from Intel team is 'The plugin works fine on iOS devices running iOS8 but iOS9 seems to not be fully supported by the plugin.'
I guess this is compatibility issue.

don, 2016-03-30:
This plugin works with iOS9. Do they mean an IntelXDK compatibility issue?

closed 2016-04-15 by don