don/cordova-plugin-ble-central #207: Not getting Notifications on IOS

redphoenix2k, 2016-03-25:
Hello,

Im subscribing to read the characteristic of a ble device using the notification command: 

ble.startNotification(BTdevice, serviceUUID, indexCharacteristic, readIndex, onError);

On android this is working well, whenever the device changes its custom 'indexCharacteristic' i immediately get my callback function fired and i can read what the new value is

But on IOS its not automatically calling the function

I tried this on two iphones 6, ios version 9, and i have the same problem

I am not getting an error or anything, simply the callback function is not getting fired/notified on change

don, 2016-03-25:
This should work the same on iOS and Android. What version of Cordova are you using? What version of the plugin?

Attach safari on OS X to your app to debug. Check the console output and make sure there aren't any JavaScript errors on iOS.

don, 2016-04-15:
@redphoenix2k Did you seen any Javascript errors in the Safari debug console? 

Can you get notifications from your BLE device on iOS using [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8)?

don, 2016-04-26:
No response, closing issue.

closed 2016-04-26 by don