don/cordova-plugin-ble-central #208: [iOS] Writing not Permitted

snowbreeze, 2016-03-28:
Error Domain=CBATTErrorDomain Code=3 "Writing is not permitted." UserInfo={NSLocalizedDescription=Writing is not permitted.}

Cordova - 6.1.0
Plugin - 1.0.4
ios - 4.1.0

Write Code
```
function writeBle(value)
{
		var device_id = connectedDevice.mac;
		var service_uuid = 'ffe0';
		var characteristic_uuid = 'ffe1';
		value = stringToBytes(value);
			
		ble.write(device_id, service_uuid, characteristic_uuid, value, function(result){	
		console.log('inside write success block');
		console.log(result);		 
		}, function(error) {		
				console.log('inside error block');
                console.log(error);            
        }) 
}

```

See Some More Info from Xcode
[image]

```
2016-03-28 17:51:13.524 plantBle[1774:824905] connect
2016-03-28 17:51:13.525 plantBle[1774:824905] Connecting to peripheral with UUID : 72A0E6AC-31F0-37E2-4820-931A36F5ED09
2016-03-28 17:51:13.704 plantBle[1774:824905] didConnectPeripheral
2016-03-28 17:51:14.093 plantBle[1774:824905] didDiscoverServices
2016-03-28 17:51:14.243 plantBle[1774:824905] didDiscoverCharacteristicsForService
2016-03-28 17:51:14.244 plantBle[1774:824905] Found characteristics for service <CBService: 0x1355c1630, isPrimary = YES, UUID = FFE0>
2016-03-28 17:51:14.245 plantBle[1774:824905] Characteristic <CBCharacteristic: 0x1355f3a30, UUID = FFE1, properties = 0x16, value = (null), notifying = NO>
2016-03-28 17:51:14.276 plantBle[1774:824905] registering for notification
2016-03-28 17:51:14.276 plantBle[1774:824905] getData
2016-03-28 17:51:14.277 plantBle[1774:824905] Looking for FFE1 with properties 16
2016-03-28 17:51:16.413 plantBle[1774:824905] stopScanTimer
2016-03-28 17:51:19.603 plantBle[1774:824905] getData
2016-03-28 17:51:19.604 plantBle[1774:824905] Looking for FFE1 with properties 8
2016-03-28 17:51:19.604 plantBle[1774:824905] Looking for FFE1
2016-03-28 17:51:19.606 plantBle[1774:824905] Error Domain=CBATTErrorDomain Code=3 "Writing is not permitted." UserInfo={NSLocalizedDescription=Writing is not permitted.}
```

snowbreeze, 2016-03-28:
writeWithoutResponse works

don, 2016-03-28:
Based on the log, I think you need you use ble.writeWithoutResponse instead of ble.write for that characteristic. 

2016-03-28 17:51:14.245 plantBle[1774:824905] Characteristic <CBCharacteristic: 0x1355f3a30, UUID = FFE1, properties = 0x16, value = (null), notifying = NO>

closed 2016-04-06 by don