don/cordova-plugin-ble-central #211: Plugin not working Galaxy S6 Android 6.0.1 with edit in config.xml, Cordova 6.1.0 and android 5.1.1

Nico924, 2016-04-06:
I saw the previous issues related to this subject however, the addition of 

<preference name="android-targetSdkVersion" value="22"/>  

in my config.xml did not change anything. I have no error but when I use the scan or startScan function, I am not able to discover any device (however if I go in the bluetooth section of my phone, I can see the device I am trying to connect to)

Do y

don, 2016-04-06:
If you're not seeing errors when starting the scan, the changes to config.xml likely worked.

When the plugin scans it finds Bluetooth Low Energy devices, if you're seeing your device in the Android Bluetooth settings, the device might be Bluetooth Classic. 

Make sure your device is advertising a Bluetooth Low Energy service. Make sure you can discover your peripheral using [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en).

don, 2016-04-15:
Duplicate of #182

closed 2016-04-15 by don