don/cordova-plugin-ble-central #212: ble.scan doesn't calls success or failure

hkharry, 2016-04-07:
refreshDeviceList: function() {
        console.log("refreshDeviceList"); // this is logged 
        // scan for all devices
              ble.scan([], 5, function(device) {
            console.log("success"); // not logged 
        console.log(JSON.stringify(device));  // not logged 
        }, function() {
        console.log("error"); // not logged 
        });
    }

don, 2016-04-07:
This sounds like the plugin isn't installed correctly or you're using Android 6.x and need to [modify config.xml](https://github.com/don/cordova-plugin-ble-central/issues/182).

In either case you should see some errors in the log file.

mohitarora24, 2016-04-09:
Similar thing is happening with me...

```
function scannow()
{
    window.plugins.toast.show('Scanning...', 'long', 'bottom', function(a){console.log('toast success: ' + a)}, function(b){alert('toast error: ' + b)}); // Works
    $("#bluetooth_outputarea").empty(); // Works
    console.log("blescanstart");
    ble.startScan([],
       function(device) {
         $("#bluetooth_outputarea").append(">>> " + device.name); // Does not happen
       },
       function(reason) {
         $("#bluetooth_outputarea").html("Scan Failed: " + reason); // Does not happen
       });
    setTimeout(ble.stopScan,
            10 * 1000,
            function() { $("#bluetooth_outputarea").append("Scan complete");
                         console.log("blescancomplete"); }, // Works
            function() { $("#bluetooth_outputarea").html("stopScan failed"); }
        );
}

ble.isEnabled(
  function() { scannow(); },  // Works
  function() {
    ble.enable(
      function() {
        window.plugins.toast.show('Bluetooth is enabled', 'long', 'bottom', function(a){console.log('toast success: ' + a)}, function(b){alert('toast error: ' + b)});  // Works
        $("#bluetooth_outputarea").html("Bluetooth is enabled"); // Works
        scannow();
      },
      function() {
        window.plugins.toast.show('The user did *not* enable Bluetooth', 'long', 'bottom', function(a){console.log('toast success: ' + a)}, function(b){alert('toast error: ' + b)});  // Works
        $("#bluetooth_outputarea").html("The user did *not* enable Bluetooth"); // Works
      }
    );
});
```

As you can see some of the ble methods are working like isEnabled and enable. startScan gets called but does not yield any scanning results when there are nearby bluetooth devices visible.

App made using Appery. Tried on OnePlus One running Android 5.1.1 and Galaxy S2 running 4.4.4. Cordova version is 4.1.1 on both (the version in Appery framework).

Output of adb logcat on S2:
```
I/chromium( 9526): [INFO:CONSOLE(173)] "blescanstart", source: file:///android_a
sset/www/libs/plugins/cordova-plugin-console/www/console-via-logger.js (173)
D/PluginManager( 9526): exec() call to unknown plugin: Console
D/BluetoothAdapterService(1109306096)( 8832): getState(): mAdapterProperties: co
m.android.bluetooth.btservice.AdapterProperties@421eb038
D/BluetoothAdapter( 9526): startLeScan(): null
D/BtGatt.GattService( 8832): registerClient() - UUID=0a64576f-4271-43b3-a805-599
962316f8b
D/BtGatt.btif( 8832): btif_gattc_register_app
D/BtGatt.btif( 8832): btgattc_handle_event: Event 1000
D/BtGatt.btif( 8832): btif_gattc_upstreams_evt: Event 0
D/BtGatt.GattService( 8832): onClientRegistered() - UUID=0a64576f-4271-43b3-a805
-599962316f8b, clientIf=5
D/BluetoothAdapter( 9526): onClientRegistered() - status=0 clientIf=5
D/BtGatt.GattService( 8832): startScan() - queue=0
D/BtGatt.GattService( 8832): startScan() - adding client=5
D/BtGatt.btif( 8832): btif_gattc_scan
D/BtGatt.btif( 8832): btgattc_handle_event: Event 1002
I/chromium( 9526): [INFO:CONSOLE(173)] "toast success: OK", source: file:///andr
oid_asset/www/libs/plugins/cordova-plugin-console/www/console-via-logger.js (173
)
D/PluginManager( 9526): exec() call to unknown plugin: Console
I/NotificationService( 2231): cancelToast pkg=io.appery.project393374 callback=a
ndroid.app.ITransientNotification$Stub$Proxy@42625e70
W/NotificationService( 2231): Toast already cancelled. pkg=io.appery.project3933
74 callback=android.app.ITransientNotification$Stub$Proxy@42625e70
D/AlarmManagerService( 2231): Kernel timezone updated to -330 minutes west of GM
T
D/BluetoothAdapter( 9526): stopLeScan()
D/BluetoothAdapterService(1109306096)( 8832): getState(): mAdapterProperties: co
m.android.bluetooth.btservice.AdapterProperties@421eb038
D/BtGatt.GattService( 8832): stopScan() - queue=1
D/BtGatt.GattService( 8832): stopScan() - queue empty; stopping scan
D/BtGatt.btif( 8832): btif_gattc_scan
D/BtGatt.btif( 8832): btgattc_handle_event: Event 1003
D/BtGatt.GattService( 8832): unregisterClient() - clientIf=5
D/BtGatt.btif( 8832): btif_gattc_unregister_app
D/BtGatt.btif( 8832): btgattc_handle_event: Event 1001
D/BtGatt.btif( 8832): btif_gattc_upstreams_evt: Event 1
W/PluginManager( 9526): THREAD WARNING: exec() call to BLE.stopScan blocked the
main thread for 31ms. Plugin should use CordovaInterface.getThreadPool().
I/chromium( 9526): [INFO:CONSOLE(173)] "blescancomplete", source: file:///androi
d_asset/www/libs/plugins/cordova-plugin-console/www/console-via-logger.js (173)
```

don, 2016-04-09:
@hkharry after reading @mohitarora24 comment I think your code might be fine. If the scan doesn't find any bluetooth devices, no callbacks will be called.

Use another app like [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to verify your phone can see other BLE devices. Your app using the plugin should see similar results.

don, 2016-04-15:
@hkharry @mohitarora24 - reopen this if you're still seeing a problem.

If your device is Android 6, try version 1.0.5 of the plugin

closed 2016-04-15 by don