don/cordova-plugin-ble-central #214: ble.disconnect wrong

fengyeezju, 2016-04-08:
I find when running in android platform 4.4.2, when i am a little far away from the ble peripheral and run the ble.disconnect, i will find  the success function in "ble.disconnect(device_id, [success], [failure]);" will be called, but the ble connection still there, and this issue happen I found i could not connect to the ble peripheral anymore unless if I manually turn off blue tooth, and re-turn on blue tooth agian. i don't know why?

don, 2016-04-08:
@fengyeezju Here's the expected behavior

When the peripheral disconnects from the phone, the failure callback from [ble.connect](https://github.com/don/cordova-plugin-ble-central#connect) should be called.

The [ble.disconnect](https://github.com/don/cordova-plugin-ble-central#disconnect) function is for the app to disconnect from the peripheral.

When a peripheral disconnects, sometimes is takes a while for the failure callback from ble.connect to be called.

Maybe the device is out of range by the time you call disconnect? On Android disconnect always returns success. https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/BLECentralPlugin.java#L216-L224

Can you get the output from `adb logcat` when the problem occurs? Maybe there's a stack trace that will point to an error. Also get any JavaScript output from chrome://inspect (assuming that works on 4.4.2)

fengyeezju, 2016-04-09:
Hi don, thank you very much for your help.but maybe something is different.

the case of mine is like this. in page 1, i called ble.connect, and then the page will be directed to page 2, since I found I can not recieve the Notification of peripherial in page2, and also i can not run code ble.startNotification once more in page2, so I need to call ble.disconnect first then ble.connect and ble.startNotification, only do this could I recieve Notification  from peripherial in page2. but the issue is when I called ble.disconnect in page2, sometimes the blue tooth connection of the phone becomes half not connected half connected(at the peripherial side it shows the peripherial was connected, but i can not send data from phone to peripherial, the only way to finish this state is to disable blue tooth at phone side and then re-turn on and re run the code ble.connect). could you kindly help me on this? thank you very much.

don, 2016-04-09:
@fengyeezju that explanation makes more sense. This plugin isn't going to work well if you're loading a new HTML page when the user navigates. When the new page loads, the plugin reloads. If you didn't disconnect properly before the new page loads, your previous connection still exists, but there's no way to get to it. 

I write my apps as single page apps, so the plugin is only ever loaded one time. When the user navigates to a new page, I show and hide divs, rather than reloading the page. This way you can keep the Bluetooth connection between pages.

Alternately you could try having a hook to disconnect before unloading the page

    window.onbeforeunload = function(e) {
        if (app && (typeof app.disconnect === 'function')) {
            console.log('Disconnecting peripheral before reload');
            app.disconnect();
        }
    };

fengyeezju, 2016-04-10:
HI DON, thank you very much.
but in the page2, the ble.write works well, only ble.disconnect have the problem.

don, 2016-04-15:
@fengyeezju It think the problem you're seeing is that plugin instance from page1 is still connected even after you disconnect from page2. Killing the app or cycling Bluetooth will disconnect that connection. I use the plugin on single page apps so I'm not seeing this.

If you can build a **simple** app that duplicates the problem reliably, I'll take a look at the source and see if there's a way to work around the problem.

whodies, 2016-06-29:
Hi, I'm having a a similar issue. If I start a connection attempt on an inactive device and turn it on midway, the resulting connection will be unbreakable. I noticed that this happens if I receive the connection callback while scanning.
Being able to stop the connection attempt before it times out could really help me out.
Thanks

closed 2016-08-15 by don