don/cordova-plugin-ble-central #215: ble not work after locked the screen on ios

wzhsh90, 2016-04-11:
I locked the screen when I scan the peripheral ,  the ble will stop to do anything . when I open the screen the ble will go on working  on stop point . How can we make it works when I locked the screen.

wzhsh90, 2016-04-11:
I find the solution.
edit project xxx-info.plist, add row  Required background modes 
and add subitems App shares data using CoreBluetooth  && App communicates using CoreBluetooth 
under Required background modes.

it works for me.

don, 2016-04-15:
@wzhsh90 I'll be looking at background scanning soon.

don, 2016-04-26:
Duplicate of #169 

See https://github.com/don/cordova-plugin-ble-central/blob/master/README.md#background-notifications-on-ios

closed 2016-04-26 by don

wzhsh90, 2016-04-27:
ok , i will use it soon.thank you.