don/cordova-plugin-ble-central #219: Can't build Android app after installed the plugin "BLECentralPlugin.java:417"

vinimendes3, 2016-04-15:
I've recently installed cordova-plugin-ble-central to my project, and it is givint to me the following error when I'm going to build or run my project to real device:

[image]

**BLECentralPlugin.java:417: error: method does not override or implement a method from a supertype
@Override
^
Note: Some input files use or override a deprecated API**

And this error does not allow me to finalize the build or run in a device.

How can I solve this? I have to continue my project...

Thanks in advance...

don, 2016-04-15:
What version of Cordova are you using?

For now, you can delete the @Override annotation to get your code to compile. (With your help I'd like to track this down and get a better fix.)

vinimendes3, 2016-04-15:
Thanks for the answer @don 

I'm using your library in another project and it is working very well ! This error does not appears during the building and running on device step. I've realized that the only difference between my current project which I'm trying to install cordova-plugin-ble-central and the another project which is working well is the version of the library. In the another project the version of the library is 1.0.4 and in my current project (which is giving the error) is version 1.0.5.

don, 2016-04-15:
Version 1.0.5 added support for Android 6.0 (API 23). Unfortunately it broke something with the older cordova.

Can you tell me what version of Cordova or PhoneGap you're running so I can try and duplicate? Run `cordova -v` or `phonegap -v`

For instance on my Mac, I get the following

    $ cordova -v
    6.1.1

vinimendes3, 2016-04-15:
I'm using version 5.4.1

don, 2016-04-15:
Until I get a fix out, you can downgrade your project to use v 1.0.4 of the plugin.

    cordova plugin rm cordova-plugin-ble-central
    cordova plugin add cordova-plugin-ble-central@1.0.4

vinimendes3, 2016-04-16:
Thanks for support @don 
I've already updated my cordova and it's working well again :D

don, 2016-04-20:
Fixed in version 1.0.6

commit 3fa264d4 (2016-04-20): fix compile issue with Cordova 5.x
fixes #219
closed 2016-04-20 by don