don/cordova-plugin-ble-central #221: can't find device after disconnect

vfistri2, 2016-04-17:
i us ble.connect and ble.disconnect when im done receiveing data, the thing is when peripheral was disconnecting after about 6 connections in a row ble.scan would stop showing peripheral in result callback. I kindof fixed it by making phone do the disconnect by calling ble.disconnect but it still sometimes for some reason stops finding device in ble.scann

vfistri2, 2016-04-17:
if device is blocked ( for uknown reason) it shows on scann after restarting phone

vfistri2, 2016-04-17:
at peripheral.java i think you should call gatt.disconnect() before gatt.close()

don, 2016-04-20:
If disconnect fails you won't be able to re-discover the device until you kill the app or restart Bluetooth.

It's possible that gatt.close() will help on some devices. Can you try this with your hardware? If it fixes the problem, send a pull request.

hervethouzard, 2016-04-21:
Hello,

Same problem for me with our own BLE peripherals and Android 5.0.1, 4.4.2 & 5.1
(no problem with iOs)

vfistri2, 2016-04-21:
have you manged to get it to bug every time? if yes how?
As far as i can tell it is on random when u disconnect on from cordova instead of from peripheral. When u disconnect from peripheral it is every 6-th time on android 4.4.2

vfistri2, 2016-04-21:
@don 
i am working on smth elese i will try to test it tommorow the thing is i am not sure what happens when it happens, and i can't get to write test case in which i know it will bug will try to add onDisconnect to call both gatt close and gatt disconnect and give users see if they stil have bugs

hervethouzard, 2016-04-21:
@vfistri2, to get it working, I have to wait 10 secondes before to be able to reconnect to the same device.

don, 2016-04-21:
@hervethouzard do you have a simple app that demonstrates the issue? That would help me test and debug on my hardware. 

Is the 10 second wait consistent across android devices and peripherals?

hervethouzard, 2016-04-22:
Hello,

The application without the peripherals is useless. Sorry.
The 10 secondes seems constant yes.

Our application is doing several exchanges (via a characteristic) with the BLE peripheral and indications are activated.

Bye

hervethouzard, 2016-04-22:
When I finish to "talk" with the peripheral, I stop notifications, I disconnect from the peripheral and stop the BLE scan.

But as I understand my logs the ble.startScan() is very long and fails.

hervethouzard, 2016-04-22:
Two more informations;

1) If I want to be sure that it will run, I have to wait 20 seconds
2) After all the process I described (stop indications, diconnect and stop scan), I have a new call to my onError callback and as long as I don't get it, the next connection will not run.

hervethouzard, 2016-04-22:
In this error callback, here is the (strange) log / error message I have:

Object {name: "SPRINGBLUE", id: "00:07:80:21:28:3D", advertising: Object, rssi: -69}
advertising
:
Object
id
:
"00:07:80:21:28:3D"
name
:
"SPRINGBLUE"
rssi
:
-69
__proto__
:
Object

hervethouzard, 2016-04-22:
I have modified my code, the last onError callback I can see is trigered by ble.connect()
If you need I have a full adb logcat available, but not publicly...

hervethouzard, 2016-04-22:
20 seconds is not the same delayon all smartphones, with a LG-D802 with Android 4.4.2 it is less than 20 seconds.

don, 2016-04-22:
@hervethouzard thanks for the info. I'll see if I can create a simple app here to duplicate. 

The JSON in the error callback for connect describes the peripheral that disconnected, so I can tell which device disconnected when I have multiple connections.

vfistri2, 2016-05-04:
@don i checked on both android devices i have 1 is sony xperia z3 compact android 5.0.1 and other is htc desire 610 android 4.4.2 ( not sure for htc which is but am sure for android version) it seems to work by calling gatt.disconnect before gatt.close. dont know how to send a pull request so can you update it? its src/android/peripheral.java at line 83 there is 		gatt.close(); just putt 		gatt.disconnect(); before it.

vfistri2, 2016-05-04:
@don ps sry that it took so long to test it was busy

hervethouzard, 2016-06-03:
Hello,

In my case the bug was coming from my code.
The ble.disconnect() call was not correct because the backuped device Id was erased by a previous code...

So no more problem for me.

Than you very much for your assistance.

don, 2016-06-05:
@hervethouzard thanks for the update

closed 2016-08-15 by don