don/cordova-plugin-ble-central #224: ble.write() Writting not permitted

AlbynoMiya, 2016-04-18:
Hey, I am trying to communicate with a BluefruitLE using an iOs application. I can scan and connect to it but I can't send any data. The error is Writing not permitted. Any ideas?

don, 2016-04-20:
The peripheral determines what you can do with the characteristic.

For some older versions of BluefruitLE you need to call `ble.writeWithoutResponse` instead of `ble.write`.

Look at the [JSON description](https://github.com/don/cordova-plugin-ble-central/#peripheral-data) of the peripheral when you connect. It will list the characteristics and permissions.

AlbynoMiya, 2016-04-20:
Thanks mate. Turns out that my BLE device doesn't support ble.write.

closed 2016-04-20 by don