don/cordova-plugin-ble-central #225: stopNotification doesn't really disable the ble notification

jj4856951, 2016-04-19:
hi:)
I don't you if it is a problem, but in my ble project, I need the android or ios to stop my ble's indication or notification.
Can set the descriptor.setValue to BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE.

don, 2016-04-20:
Both iOS and Android make API calls to stop the notification

https://github.com/don/cordova-plugin-ble-central/blob/4e1394cb9f5970460c38e7b4470546b33b1f5c2f/src/ios/BLECentralPlugin.m#L199

https://github.com/don/cordova-plugin-ble-central/blob/b6fda752eb8d1e828306bd2e35fce53e3391ceb0/src/android/Peripheral.java#L341

It's highly possible Android gatt.setCharacteristicNotification is useless and the descriptor also needs to be set.

Where are seeing problem? (Platform, OS version, Hardware)
Do you have a sample app that demonstrates this?

jj4856951, 2016-04-26:
Thank you for your answer, but I am still on the problem. 
When I use 'ble.stopNotification()' in my ionic app on Android, my app can stop listen, but my ble device is still sending data. So, the method didn't stop the ble.

This is the Android phone's bluetooth hci_log:
[android_phone_hci_log.zip](https://github.com/don/cordova-plugin-ble-central/files/236192/android_phone_hci_log.zip)

The "...59.log" is the right one (a native app which can stop the ble's notification );
`
Frame 760: 14 bytes on wire (112 bits), 14 bytes captured (112 bits)
Bluetooth
Bluetooth HCI H4
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth Attribute Protocol
    Opcode: Write Request (0x12)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 0010 = Method: Write Request (0x12)
    Handle: 0x004f (Client Characteristic Configuration)
        [UUID: Client Characteristic Configuration (0x2902)]
    Characteristic Configuration Client: 0x0000
        0000 0000 0000 00.. = Reseved: 0x0000
        .... .... .... ..0. = Indication: False
**        .... .... .... ...0 = Notification: False**
`
The "...11.log" is the ionic app with this plugin, has the problem.
`
Frame 487: 14 bytes on wire (112 bits), 14 bytes captured (112 bits)
Bluetooth
Bluetooth HCI H4
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth Attribute Protocol
    Opcode: Write Request (0x12)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 0010 = Method: Write Request (0x12)
    Handle: 0x004f (Client Characteristic Configuration)
        [UUID: Client Characteristic Configuration (0x2902)]
    Characteristic Configuration Client: 0x0001, Notification
        0000 0000 0000 00.. = Reseved: 0x0000
        .... .... .... ..0. = Indication: False
**        .... .... .... ...1 = Notification: True**
`

jj4856951, 2016-04-26:
ps: ios doesn't have the problem.

jj4856951, 2016-04-27:
Problem solved!
just add:
>
 BluetoothGattDescriptor descriptor = characteristic.getDescriptor(CLIENT_CHARACTERISTIC_CONFIGURATION_UUID);
descriptor.setValue(BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE);
gatt.writeDescriptor(descriptor);

after line 341 in android's peripheral.java.
Thank you:)

closed 2016-08-19 by don

markodayan, 2017-10-23:
Does anyone know how to fix this within Ionic?

markodayan, 2017-10-23:
^ My android app doesnt stop the notifications when I am logging out a user profile hence when I log in again, the notifications are served in doubles.

ddugue, 2017-10-23:
hey @markodayan ,
Did you call the stopNotification call before disconnecting?

If so, it might be linked with #481, because the command queue is sometimes not cleared up properly

commit 99d5d6b0 (2016-08-19): stopNotification writes DISABLE_NOTIFICATION_VALUE to the descriptor
fixes #225