don/cordova-plugin-ble-central #229: "Undefined" BLEPeripheral Name

lauleehong, 2016-04-28:
I'm getting "undefined as my BLE-peripheral name on my Android. Any idea how I will be able to detect it like in iOS? I'm able to get the device name of the same hardware on your Cordova SimpleSerial iOS. 

I added:  console.log(JSON.stringify(device));        //check peripheral info packet 
inside onDiscoveryDevice.
"{"id":"AA:BB:CC:DD:EE:FF","advertising":{},"rssi":-55}"    // real id masked; missing "name"

don, 2016-04-29:
Some devices don't advertise a name.

Once you're connected you could try reading the [Device Name Characteristic 0x2A00](https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicViewer.aspx?u=org.bluetooth.characteristic.gap.device_name.xml) from the [Generic Access Service 0x1800](https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.generic_access.xml)

I think Android looks for the Complete Local Name 0x09 from the advertised [GAP information](https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile). You could try parsing the advertising data from the peripheral and seeing if 0x09 exists or perhaps 0x08 Shortened Local Name.

Does [NRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) show the name correctly?

lauleehong, 2016-04-29:
How should I go about reading the device name characteristics on cordova/js? Spot on that my Complete Local Name is "N/A" and MCP on Android shows "N/A" as well (but shows Full Name on iOS).

don, 2016-05-03:
You'll need to connect to the peripheral and do something like

    var success = function(buffer) {
        var name = ''; // convert bytes in ArrayBuffer to a string
        console.log('device name is', name); 
    };
    ble.read(peripheral.id, '1800', '2a00', success, app.onError);

lauleehong, 2016-05-04:
I'm working on a BLE scanner list, so to sequentially connect each peripheral to parse out the device name seems impractical.

don, 2016-05-05:
If you don't want to connect, I'd use N/A or the MAC address for the name.

lauleehong, 2016-05-05:
If I were to populate a list with BLE peripherals, an average joe wouldn't know which device he is looking for to connect to if the name was N/A or the MAC address.

don, 2016-05-05:
If the device doesn't advertise a name, it's not going to be in the scan results on Android. iOS and Android handle this differently.

nRF Master Control Panel is a generic scanning app. It shows N/A for devices that don't advertise names. Most devices that are connectable advertise a name.

My plugin provides access to the Android and iOS API. I can't give you data that's not there. You need to figure out what makes sense for your app. Maybe N/A is good. Maybe you need to connect and read the device name if it's not advertised.

lauleehong, 2016-05-05:
My hardware does advertise a name, and thus I'm able to pick it up on my iPhone. I'm using your plugin to make a "remote controller" like app which list the names of the devices at start. The user then selects which device to connect to for UART-BLE data transmission.

don, 2016-05-05:
Based on what you've told me, the device name you see on iOS is coming from the Generic Access Service 0x1800, Device Name Characteristic 0x2a00. On iOS, CoreBluetooth gets the name internally, and makes it available as peripherals are discovered. It is probably by doing a Read By UUID Request (but this API isn't exposed.) CoreBluetooth does other stuff during discovery like hiding the MAC address and generating a UUID for the peripheral.

On Android the device name needs to be in the advertising or scan response packet. 

Can you update the firmware on the device? I think you'll get the best results if you advertise the local name **and** set the device name.

lauleehong, 2016-05-05:
Don, thanks for clarifying. From my understanding, a device name would not be short enough to fit into a packet if it was already advertising NID and BID through Eddystone. I have encoded 2 bytes in my eddystone NID as a "device name" identifier. I plan to fish it out from the Generic Access Profile 0x16.Am I able to access that through your plugin in anyway?

don, 2016-05-05:
Remember you can send 31 additional bytes in the scan response packet. It works like the advertising info. I'd add the device name there. 

You can get access to the Eddystone data from the plugin, however it's a bit manual right now. I plan to add helper functions in a future version (see #192). Until then, here's an example of [creating a hash map from the advertising data](https://github.com/don/ITP-BluetoothLE/blob/887511c375b1ab2fbef3afe210d6a6b7db44cee9/phonegap/thermometer_v2/www/js/index.js#L18-L39) and [parsing out some service data](https://github.com/don/ITP-BluetoothLE/blob/887511c375b1ab2fbef3afe210d6a6b7db44cee9/phonegap/thermometer_v2/www/js/index.js#L93-L103).

lauleehong, 2016-05-06:
Is the hash map function mentioned generic and ready to be used as it is or I would have to change certain parameters?

don, 2016-05-06:
I'd do something like this 

    var SERVICE_DATA_KEY = '0x16';
    var advertisingData = parseAdvertisingData(device.advertising);
    serviceData = advertisingData[SERVICE_DATA_KEY];
    if (serviceData) {
        // first 2 bytes are the 16 bit UUID
        var uuidBytes = new Uint16Array(serviceData.slice(0,2));
        var uuid = uuidBytes[0].toString(16); // hex string
        console.log("Found service data for " + uuid); // should be 0xFEAA

        // remaining bytes are the Eddystone beacon data
        var eddystoneData = new Uint8Array(serviceData.slice(2));
        // do something with the eddystoneData
    }

closed 2016-05-06 by lauleehong