don/cordova-plugin-ble-central #231: Add browser support

heukirne, 2016-05-01:
https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web

don, 2017-01-04:
[WebBluetooth](https://googlechrome.github.io/samples/web-bluetooth/) works well on a lot of devices running Chrome. Unfortunately the API is a bit different so it's not trivial to add support  here.

Hollerweger, 2017-01-04:
Do you mean that in general the WebBlutooth API is is too different to
Android and iOS native APIs for a Cordova Plugin or was the comment ony
specific to your cordova plugin API.

don, 2017-01-04:
Currently web bluetooth API is too different from this API to *easily* create a wrapper.  It's also Chrome only, so for now I think it's best to keep them separate. If/when the API is finalized, I'll probably support that API. Until then it's going to be really low priority until a client needs it.

FWIW, the WebBluetooth API works pretty well and is a good choice where you can target Chrome on OS X, Linux, ChromeOS and Android. Unfortunately it doesn't work with Chrome on Windows or iOS yet.

Hollerweger, 2017-01-28:
Chrome stable 56 with WebBluetooth is now released

Tjieco, 2018-05-28:
Any update on this?

don, 2018-06-07:
WebBluetooth is a good API but it works different from this plugin and it's Android only.

I have no immediate plans to implement WebBluetooth. If anyone needs this for a project, email me and we can talk about implementing it for you on a contract basis.

mpacary, 2019-03-27:
For anyone interested in using a cordova bluetooth plugin through Web bluetooth API, check [this project](https://github.com/tzachari/cordova-plugin-webbluetooth/blob/master/webbluetooth.js)
I could get it work for my needs by tweaking 2-3 small things.
Note that the underlying bluetooth plugin is not cordova-plugin-ble-central.

ryios, 2019-05-16:
I don't deploy to the browser, I just run locally to test on the browser.  So what I do is write a wrapper over bluetooth and if it's the browser build it uses chromes blue tooth api.

I use my wrapper as an abstraction layer into either so my code will work on Browser, Windows, Android, or IOS.

sonnenhohl, 2019-05-16:
@ryios from what i understand you created a layer over cordova-plugin-ble-central and WebBluetooth and according to the platform you use the specific code.
You created that for testing because of performance of deploying into Cordova?
Anyway i think your solution is very interesting in my case i would be interested in using something like that to also being able to deploy our app to the web, but for that i will need to wrap other apis also.

ryios, 2019-05-16:
Yeah, I use live reload locally when developing so I can make live changes to my JS/CSS/Html etc and see it refresh in the browser and so I can use browser debugging tools and developer console.

Windows is the only platform I could test blue tooth on locally without having to due long tedius deployments and I didn't have a developer console which annoyed me.

So basically I just wrote an abstraction layer into bluetooth and when running in the browser it uses chromes blue tooth api, otherwise it uses the Cordova Plugin.  There are some differences in restrictions between the two apis I'm still ironing out, but the basic tests I've done so far work.

I don't how viable it would be for a browser deployment, it would definitely only work in Chrome.

hthetiot, 2019-08-02:
Resources:
- https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web
- https://googlechrome.github.io/samples/web-bluetooth/

Initial draft:

```
function notSupported() {
    console.log('BLE is not supported on the browser');
}

function log(msg) {
    console.log(msg);
}

module.exports = {

    devices: new Map(),
    scanning: null,

    scan: function(services, seconds, success, failure) {
        notSupported();
        if (failure) failure();
    },
    startScan: function(services, success, failure) {
        this.scanning = true;
        navigator.bluetooth.requestDevice({
            "acceptAllDevices": true,
            filters: [{
                services: services
            }]
        }).then(device => {
            if (this.scanning) {                
                success(device);
            }
        }).catch(failure);
    },
    stopScan: function(success, failure) {
        this.scanning = false;
        if (success) success();
    },
    startScanWithOptions: function(services, options, success, failure) {
        notSupported();
        if (failure) failure();
    },
    connect: function(device_id, connectSuccess, connectFailure) {
        navigator.bluetooth.requestDevice({filters: [{
            services: [device_id]
        }]}).then(device => {
            log('Connecting to GATT Server...');
            return device.gatt.connect().then(server => {
                log('Getting Device...');
                this.devices.set(device_id, {
                    device: device,
                    server: server
                })
                connectSuccess(device);
            });
        }).catch(connectFailure);
    },
    disconnect: function(device_id, disconnectSuccess, disconnectFailure) {
        if (this.devices.has(device_id)) {
            this.devices.get(device_id).server.gatt.disconnect().then(result => {
                this.devices.delete(device_id);
                return result;
            }).then(disconnectSuccess, disconnectFailure);
        } else if (connectFailure) {
            connectFailure();
        }
    },
[... 86 lines cut ...]
        if (this.devices.has(device_id)) {
            success();
        } else if (failure) {
            failure();
        }
    },
    showBluetoothSettings: function(success, failure) {
        notSupported();
        if (failure) failure();
    },
    enable: function(success, failure) {
        notSupported();
        if (failure) failure();
    },
    startStateNotifications: function(success, failure) {
      notSupported();
      if (failure) failure();
    },
    stopStateNotifications: function(success, failure) {
      notSupported();
      if (failure) failure();
    }
};

```

peitschie, 2022-03-14:
Resolved via #907

closed 2022-03-14 by peitschie