don/cordova-plugin-ble-central #232: ble.scan not working on Nexus 5

vinimendes3, 2016-05-02:
I'm trying to find visible devices using the same code in Nexus 5 and LG G3. 

LG G3 device is working well, finding visible devices normally, but Nexus 5 is not finding any device.

Why the same code has differently behavior in two devices?

Thanks in advance!

vinimendes3, 2016-05-03:
I've also tried in **Sony Xperia M2 Dual (C2004)** and not worked. 

I'm concerned because I need as soon as possible to solve this =(

don, 2016-05-03:
You'll see different behavior on different hardware and different versions of Android. In my experience, filtering scan results by service UUIDs on older versions of Android is particularly bad. 

For some applications, you might need to scan for devices with any service and process the scan results manually to find the peripherals you care about.

vinimendes3, 2016-05-03:
I'm scanning without specify service UUID. I'm passing an empty array as parameter in scan calling. My code loooks like this:

		$cordovaBLE.scan([], 5)

			.then(

				function(response) {

				},

				function(error) {

				},

				function(device) {

				}
			);

In LG G3 works well and return to me all devices around. But in other devices it not works.

don, 2016-05-03:
What versions of Android are causing problems?

Do you see your device when scanning with [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en)?

It's possible the Bluetooth stack crashed. Try rebooting the device (or restarting wifi and Bluetooth.) 

Do you get the same results without the ngcordova wrapper?  

Also, if one phone is connected to the peripheral, it will stop advertising and the other phones won't be able to find it.

vinimendes3, 2016-05-03:
My LG G3 is running Android 6.0 and working well.
Nexus 5 is also running Android 6.0
Sony Xperia M2 is running Android 4.3

vinimendes3, 2016-05-03:
In Nexus5 and Sony Xperia when I scan through nRF Master Control Panel it finds the peripheral normally, but working programatically it does not find these peripherals.

don, 2016-05-03:
I have a Nexus 5 around here somewhere. I'll see if I can duplicate. Does your peripheral have 16 bit or 128 bit UUID?

vinimendes3, 2016-05-03:
Scanning on Nexus 5, this is my peripheral:

[image]

vinimendes3, 2016-05-03:
Now, I don't know why, Nexus 5 has found the peripheral '-'. 

But Sony Xperia M2 Dual still not find.

I can see peripheral in bluetooth settings of Sony Xperia, but when I try to scan via nRF Master, it returns me the following situation when I open the app:

[image]

And searching via Bluetooth native Settings, it founds normally:

[image]

don, 2016-05-03:
I see the problem now. The Sony Xperia M2 Dual doesn't support Bluetooth Low Energy (aka Bluetooth Smart.) Android 4.3 added software support for BLE, but you also need Bluetooth 4.0 hardware.

vinimendes3, 2016-05-03:
After which Android version was added this compatibility? Just to set this version as minimum version in my app.

don, 2016-05-03:
It was added in Android 4.3 (API 19)

To restrict your app to BLE-capable devices only, include the following in your app's manifest:

    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

vinimendes3, 2016-05-04:
Thanks @don :D 
But is it normal to appears the peripheral via Bluetooth Settings scan and not appears via library scan? 
I ask this because Sony Xperia M2 is running Android 4.3, and is not showing the peripheral on scan.

don, 2016-05-04:
Yes, I think thats normal. See the nRF master control panel screenshot. It says the device doesn't support BLE. Unfortunately you probably have a Bluetooth 2.1 or 3.0 chipset in that device.

vinimendes3, 2016-05-04:
If I add:

    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

My app will allow only devices compatible with Bluetooth BLE having Bluetooth 2.1, 3.0 and so on to install my app? Or even does it have a chance to take a device not compatible with Bluetooth 2.1 and 3.0 like Sony Xperia?

don, 2016-05-27:
@vinimendes3 if users MUST have Bluetooth Low Energy add the following line to the manifest

    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

The Play store should restrict to only compatible devices.

closed 2016-05-27 by don