don/cordova-plugin-ble-central #233: startNotification "BluetoothGatt is null" error on Android 6.0

artman186, 2016-05-03:
I have a device (Motorola XT1585, Droid Turbo 2) that was recently updated to Android 6.0 from Android 5.0.  On this device I'm able to make a connection to a BLE device (a scale, in this case) but when I call "startNotification" the device immediately disconnects and then invokes the startNotification failure callback with the message "BluetoothGatt is null".  The exact same code is working fine on a Asus K01A running Android 5.0.  I was previously running version 0.1.3 of the plugin, I upgraded to 1.0.6 and am experiencing the same issue.

don, 2016-05-03:
Can you run `adb logcat` and see if there is a stack trace?

artman186, 2016-05-03:
The Android logs don't indicate any errors, but I do have the adb output from the device while this was happening.  I've sent what I have via e-mail, let me know what else I can provide.

don, 2016-05-27:
@artman186 you probably need to upgrade to the latest version of the plugin with new Android-6 permission support. You need at least version 1.0.5, but I recommend using the latest.

    cordova plugin rm cordova-plugin-ble-central
    cordova plugin add cordova-plugin-ble-central

artman186, 2016-05-28:
We are using this plugin through the verified Telerik plugins marketplace, and although their versioning scheme is a little goofy, according to them we are using version "1.1.0.2", although the only versions they show are 0.1.3 and 1.1.0.2 depending on which part of the tool you are in.

Regardless, I just re-tested this issue on Android 6.0 using Telerik's "1.1.0.2" version of the plugin and everything worked correctly, so I will mark this closed.

closed 2016-05-28 by artman186

don, 2016-05-28:
@artman186 Is the Telerik v1.1.0.2 different from my 1.1.0? Any way you can test with my version?