don/cordova-plugin-ble-central #234: MTU limit of 20 bytes on some Android devices

airbly, 2016-05-03:
Hi,

I am not sure how to re-open an issue.  This is in reference to issue  #55 

I am having issues writing more than 20 bytes at a time on Android 4.4.2 as well.

It appears that you can direct Android to negotiate an MTU as it does not automatically do so.

http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestMtu%28int%29
public boolean requestMtu (int mtu)

Is this something you can consider adding to this API? ble.requestMTU(size) or something to that effect?

Thanks for all your help otherwise. This library is very handy.

Chris

don, 2016-05-03:
@airbly I'm not sure that increasing the MTU will help. It's my understanding that for characteristics exceeding the MTU, Android (and iOS) breaks the data into chunks and sends multiple blob packets. (I should verify this with a sniffer, but haven't.)

Is this just an Android 4.4.2 issue? (Android 4.3 and 4.4 are finicky.)

Can you try modifying the your version of the plugin to see if it actually works? Try something simple like requestMtu in [connect](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L65) and add [onMtuChanged](http://developer.android.com/reference/android/bluetooth/BluetoothGattCallback.html#onMtuChanged(android.bluetooth.BluetoothGatt,%20int,%20int)) so you can log success/failure.

airbly, 2016-05-03:
Hi Don,

I can confirm that on iOS, when the MTU on my peripheral is set to 128 for example, the BLE messages are sent in 128 byte chunks.  I believe this is because iOS automatically negotiates the MTU as where Android (at least 4.4.2) does not.

The problem this is causing me is when I upgrade the firmware on my peripheral using Android, it takes approximately 6 times as long to do in 20 byte chunks.

I have not tried to alter your library.  I will see if I can figure out how to do so and let you know how I make out. 

Thanks for your quick reply.

- Chris

hervethouzard, 2016-05-04:
Hello,

It's not Android specific, it's the norm.
20 bytes is the maximum, you have to split your data.

airbly, 2016-05-09:
Hi,

The 20 / 23 byte limit is not actually a limit.  It is the default in the specification.  There is nothing in the BLE or the L2CAP specification that limits the client and peripheral from negotiation a different MTU.  In fact there is an Exchange MTU procedure designed to do exactly that.  

iOS devices appear to negotiate an MTU automatically and have no problems communicating in 140 byte MTU packets (what I am currently testing).  All iOS communication with that MTU setting will communicate in single packets without fragmentation up to that size.

Android (4.4.2 at least) however does not appear to automatically negotiate that MTU size and the client must request the MTU negotiation as I understand it.  

- Chris

odbol, 2016-05-19:
Android clients can definitely negotiate the MTU size to be bigger than the default 23 bytes. However, I've never gotten it to work. Seems to depend on the phone you're using.

airbly, 2016-10-06:
Hi Don,

Do you have any instructions to setup the build environment to recompile your library?  Or can you point me to a good howto somewhere on the internet?  I've never done this before but I would like to take a stab at creating the code to cause Android to send an MTU request after connection.

I've confirmed that iOS 9 automatically requests the MTU after it connects to a peripheral.  Android does not do this automatically however.

Thanks.

- Chris

loopezz, 2017-03-15:
Hi,

Currently I can't send more than 20 bytes of data. How can I extend that to 1029 bytes? The peripheral does support it. Adding gatt.requestMtu(1032); on [line 76](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L76) doesn't increase it on Android 6.0.1 (API 23).  However, I can send 1029 bytes of data on iOS. Any ideas for an implementation of it to the plugin?

Thanks in advance.

don, 2017-03-15:
The max characteristic size is 512 bytes for Bluetooth 4 assuming your hardware supports it. Large characteristics are written using BLOBs and offsets. iOS and Android handle this transparently. Increasing the MTU means bigger chunks are used there fewer these BLOB writes. A MTU of 1032 isn't going to work.

I can write 500 byte characteristics to OS X from iOS and Android without changing any code in this plugin. On many Android phones, I've seen lower limits.

If you really need to write 1029 bytes you'll need to do multiple writes or split the data across multiple characteristics.

loopezz, 2017-03-16:
Well, I notice some difference between iOS and Android. With iOS (10.2.1) I can send an ArrayBuffer with a length of 1029 (bytes), but Android (6.0.1) does send only 20 bytes of it. Why doesn't Android handle this?

develop-algoritma, 2017-09-18:
Hi,
i had the same problem. My BLE Device (custom device) need at leas 32 bytes per packet.

So i made a fork of the project and add requestMTU() function (Android 5>).

Hope it can help someone... 

This is the fork: https://github.com/develop-algoritma/cordova-plugin-ble-central

anjildhamala, 2017-10-27:
@develop-algoritma Have you tried to use your fork in ionic?

develop-algoritma, 2017-10-30:
Yes,
we use it on Ionic (1.3) but without Ionic Native. We develop an angular service that encapsulate plugin functions and exposes them through promises.

Hope we help you,
Bye

anjildhamala, 2017-10-31:
if anybody stumbles across this page, I hope this implementation will be useful for you. I was able to set higher mtu with this small snippet of code. (tested in ionic 3)

```
private void requestMtu(final CallbackContext callbackContext, String macAddress, int mtuValue) {
        Peripheral peripheral = peripherals.get(macAddress);

        BluetoothGattCallback bleCallBack = new BluetoothGattCallback() {
            @Override
            public void onMtuChanged(BluetoothGatt gatt, int mtu, int status) {
                super.onMtuChanged(gatt, mtu, status);
                LOG.d(TAG, "mtu changed");
                callbackContext.success();
            }
        };

        BluetoothDevice device = peripheral.getDevice();
        BluetoothGatt bluetoothGatt = device.connectGatt(cordova.getActivity().getApplicationContext(), true, bleCallBack);

        LOG.d(TAG, "request MTU peripheral exists " + macAddress);
        bluetoothGatt.requestMtu(mtuValue);
    }
```
Add this method in BLECentralPlugin.java.

nicocg91, 2017-11-03:
Hi everyone

@anjildhamala  thanks for your help. It worked for me! But I would like to say that it was necessary for me adding a delay before:

bluetoothGatt.requestMtu(mtuValue);

If not, callback bleCallBack doesn't work. Do you know the reason?

anjildhamala, 2017-11-03:
@nicocg91 As beautifully as it worked the first few times, the above code did not resolve anymore since yesterday and I had to resort to something else. I went back to @develop-algoritma's fork and then added the necessary code in Peripheral class's requestMtu() function. I resolve from BLECentralPlugin class's requestMtu() function on the true value being returned from the requestMtu() method of Peripheral class and error if it returns false. My assumption is that it only returns false if the current android version of the device in use doesn't support mtu change. 

Then I have a delay set up in my javascript. My work involved changing mtu, establishing subscription and then requesting a few things that would come through the subscription. I request for the mtu and then wait a second before establishing subscription and doing everything else. If I do not wait for it, the onMtuChanged function never gets called. Might have to do with bluetooth overwriting whatever the previous command command was and moving on to the newer command as it resolves pretty rapidly if the device does support mtu change. 

The above snippet is sorta different. it only resolves on the callback and hence, it should be bullet proof as long as you are waiting for it to resolve and do other bluetooth stuff.

develop-algoritma, 2017-11-03:
@nicocg91 , @anjildhamala ,
The fork already do what you want, but the requestMTU function must be  called after connect succeed (in the success callback). here a snippet of code that use plugin to connect to a device (device_id) and after that , set MTU size:

Hope we help you,
Bye

      ble.connect(device_id,
        function (result) {
          if (result != null) {
            ble.requestMtu(device_id, mtuValue,
              function (result) {
                // resolve and continue with your code
                resolve(result);

              },
              function (error) {
                reject(error)
              }
            );
          } else
            reject("connection error");
        },
        function (error) {
          reject(error)
        }
      );

anjildhamala, 2017-11-03:
@develop-algoritma : Yeah, I would say that the fork does work. Only thing I'd change is return the boolean response from the Peripheral's requestMtu() function and resolve only if you get true and error when you get false. 
Regardless, we do have to set a delay to make the mtu change stick if we want to immediately do something after we change mtu. something like 
connect => changeMtu => doSomeBLEStuff
 needs a delay after changeMtu

odbol, 2017-11-05:
Be warned that calling changeMtu() crashes the bluetooth stack on some older (Marshmallow) phones: https://issuetracker.google.com/issues/37101017

airbly, 2018-02-27:
Hi Don,

Do you have any plans to implement the requestMtu function from @develop-algoritma 's fork of your code?  A larger negotiated MTU will lead to significant performance improvements.  iOS devices appear to automatically try to negotiate the highest MTU possible.  For example:

If my device limits the MTU to 23 bytes, transferring 512kB of data via notifications takes about 5 minutes on an iOS device.  If I then increase the allowed MTU of my device to 140 and allowing the iOS device to negotiate that higher MTU, that transfer time drops to under 1 minute.

Most Android devices do not automatically negotiate a higher MTU.  It would be help improve performance if that requestMtu option were implemented thus allowing more data to be sent during the BLE transmit/receive intervals.

Thanks.

- Chris

odbol, 2018-02-27:
Keep in mind that requestMtu isn't widely supported, and instead of gracefully failing it will crash the entire Bluetooth stack on some phones. I've found that Samsung are the only ones that reliably support it on older phones.

airbly, 2018-03-02:
Do we know how the app crashes when it is called on an older phone?  Is it  an exception that just needs to be caught?

odbol, 2018-03-03:
No exception is thrown. The whole Bluetooth stack goes down from a crash in native code. Best you'll get is an error message in logcat, and a completely unpredictable, unrecoverable broken Bluetooth where you have to restart the whole phone.

odbol, 2018-03-03:
Here's is the bug I filed. Mighty be fixed in newer Android versions. https://issuetracker.google.com/u/1/issues/37101017

mirkomancin, 2018-03-15:
@develop-algoritma @odbol How can I fill out my custom version of the plugin? I writed the code you said but do not fill out the form inside the "node_modules" folder

yuenkhang, 2018-04-08:
@nicocg91 ,
Hi I'm just a student working on this mtu and I found this page is really useful. Regarding the delay issue, my solution is wait for the onMtuChanged() event to fire before you continue with other work.

don, 2018-05-10:
Fixed in #568

closed 2018-05-10 by don