don/cordova-plugin-ble-central #235: Unhandled exception crash on connect

artman186, 2016-05-06:
Hey Don, got another issue for you.  We have a sporadic error occurring that actually creates a hard crash of the application (this doesn't happen every time, but does happen often).  The iOS crash logs aren't much help, although I have attached one (this happens on multiple devices).  I think the key lies in the JavaScript console output right before the app crashes:

<Warning>: connect
<Warning>: Connecting to peripheral with UUID : E85105F1-8FDE-DEEC-6136-136D443828A1
<Warning>: didConnectPeripheral
<Warning>: didDiscoverServices
<Warning>: didDiscoverCharacteristicsForService
<Warning>: Found characteristics for service <CBService: 0x16148d0a0, isPrimary = YES, UUID = 1BC50001-0200-0AA5-E311-24CB004A98C5>
<Warning>: Characteristic <CBCharacteristic: 0x15f7acd20, UUID = 1BC50002-0200-0AA5-E311-24CB004A98C5, properties = 0x12, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x15f76e7e0, UUID = 1BC50006-0200-0AA5-E311-24CB004A98C5, properties = 0x12, value = (null), notifying = NO>
<Warning>: didDiscoverCharacteristicsForService
<Warning>: Found characteristics for service <CBService: 0x16142e6c0, isPrimary = YES, UUID = B5600001-A0F8-16AF-BB42-1D3B642EC2E1>
<Warning>: Characteristic <CBCharacteristic: 0x1613395d0, UUID = B5600003-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0x12, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161339630, UUID = B5600004-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0x12, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x15f7d90f0, UUID = B5600005-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0xA, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x15f7d9150, UUID = B5600007-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0x12, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161392cd0, UUID = B5600008-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0xA, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161392d30, UUID = B5600009-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0xA, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161392d90, UUID = B560000B-A0F8-16AF-BB42-1D3B642EC2E1, properties = 0x8, value = (null), notifying = NO>
<Warning>: didDiscoverCharacteristicsForService
<Warning>: Found characteristics for service <CBService: 0x16148c300, isPrimary = YES, UUID = Battery>
<Warning>: Characteristic <CBCharacteristic: 0x1613304f0, UUID = Battery Level, properties = 0x12, value = (null), notifying = NO>
<Warning>: didDiscoverCharacteristicsForService
<Warning>: Found characteristics for service <CBService: 0x1614901b0, isPrimary = YES, UUID = Device Information>
<Warning>: Characteristic <CBCharacteristic: 0x16136da20, UUID = Manufacturer Name String, properties = 0x2, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x16138af30, UUID = Model Number String, properties = 0x2, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161347560, UUID = Hardware Revision String, properties = 0x2, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161355880, UUID = Firmware Revision String, properties = 0x2, value = (null), notifying = NO>
<Warning>: didDiscoverCharacteristicsForService
<Warning>: Found characteristics for service <CBService: 0x15f5f7730, isPrimary = YES, UUID = 00001530-1212-EFDE-1523-785FEABCD123>
<Warning>: Characteristic <CBCharacteristic: 0x16137ac50, UUID = 00001532-1212-EFDE-1523-785FEABCD123, properties = 0x4, value = (null), notifying = NO>
<Warning>: Characteristic <CBCharacteristic: 0x161375770, UUID = 00001531-1212-EFDE-1523-785FEABCD123, properties = 0x18, value = (null), notifying = NO>
<Warning>: read
<Warning>: getData
<Error>: -[NSNull isEqualToString:]: unrecognized selector sent to instance 0x1a1af7490
<Error>: *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSNull isEqualToString:]: unrecognized selector sent to instance 0x1a1af7490'

We are running version 1.1.0, this is happening on various iPads running version 9.2.x and 9.3.1.  The attached crash logs are from an iPad Air 2 running iOS 9.3.1.
[crashes.zip](https://github.com/don/cordova-plugin-ble-central/files/252987/crashes.zip)

don, 2016-05-06:
Based on the console output, it looks like the error happens in [getData](https://github.com/don/cordova-plugin-ble-central/blob/f7bc0b5313ea7e06915e1538c1893aad9032f901/src/ios/BLECentralPlugin.m#L691-L770) after [read](https://github.com/don/cordova-plugin-ble-central/blob/f7bc0b5313ea7e06915e1538c1893aad9032f901/src/ios/BLECentralPlugin.m#L105-L120) is called.

Maybe a Null is being passed to [findPeripheralByUUID](https://github.com/don/cordova-plugin-ble-central/blob/f7bc0b5313ea7e06915e1538c1893aad9032f901/src/ios/BLECentralPlugin.m#L619-L633)?

It's going to be tough to duplicate here. I'll see if I can add some better Null handing and logging to help track it down.

artman186, 2016-05-09:
I added a lot more logging myself, and it turns out the issue was on my end.  We identified a scenario where we were in fact sending an "undefined" as a device ID so the crash wasn't as "inconsistent" as we thought.  My apologies, thanks for checking into this.

don, 2016-05-09:
Thanks for the update. I'm going to reopen the issue until I add a better error message for this case in the plugin.

don, 2018-06-07:
Both iOS and Android call the error callback now if a bad id is passed to connect

closed 2018-06-07 by don