don/cordova-plugin-ble-central #237: Error building from phonegap

grisotti, 2016-05-09:
hi,
when I try to build alwais receive this error:

Error - Some official plugins have to be updated if using PhoneGap >= 4.0.0. Please upgrade the version of any plugins that may include the following file: Peripheral.java

here the log:

Build Date: 2016-05-09 07:55:19 +0000
--------------------------------------------------------------------------------
PLUGIN OUTPUT
--------------------------------------------------------------------------------
Fetching plugin "cordova-plugin-ble-central" via npm
Installing "cordova-plugin-ble-central" at "1.1.0" for android
Fetching plugin "cordova-plugin-compat" via npm
Installing "cordova-plugin-compat" at "1.0.0" for android
--------------------------------------------------------------------------------
COMPILE OUTPUT
--------------------------------------------------------------------------------
Executing: ant debug -f /project/build.xml
Buildfile: /project/build.xml
Buildfile: /project/build.xml

-check-env:
 [checkenv] Android SDK Tools Revision 24.4.1

-setup:
     [echo] Project Name: .IoTSemplice

-set-mode-check:

-set-debug-files:

-check-env:
 [checkenv] Android SDK Tools Revision 24.4.1

-setup:
     [echo] Project Name: .IoTSemplice
  [gettype] Project Type: Application
  [gettype] Project Type: Application

-pre-clean:

clean:

-set-debug-mode:

-debug-obfuscation-check:

-pre-build:

-build-setup:
[getlibpath] Library dependencies:
[getlibpath] 
[getlibpath] ------------------
[getlibpath] Ordered libraries:
[getbuildtools] Using latest Build Tools: 23.0.1
     [echo] Resolving Build Target for .IoTSemplice...

nodeps:

-check-env:
 [checkenv] Android SDK Tools Revision 24.4.1

-setup:
     [echo] Project Name: .IoTSemplice
  [gettype] Project Type: Android Library

-pre-clean:

clean:

BUILD SUCCESSFUL
Total time: 1 second
[gettarget] Project Target:   Android 5.1.1
[gettarget] API level:        22
     [echo] ----------
     [echo] Creating output directories if needed...
    [mkdir] Created dir: /project/bin
    [mkdir] Created dir: /project/bin/res
    [mkdir] Created dir: /project/bin/rsObj
    [mkdir] Created dir: /project/bin/rsLibs
    [mkdir] Created dir: /project/gen
    [mkdir] Created dir: /project/bin/classes
    [mkdir] Created dir: /project/bin/dexedLibs
     [echo] ----------
     [echo] Resolving Dependencies for .IoTSemplice...
[dependency] Library dependencies:
[dependency] 
[dependency] ------------------
[dependency] Ordered libraries:
[dependency] 
[dependency] ------------------
     [echo] ----------
     [echo] Building Libraries with 'debug'...

nodeps:

-set-mode-check:

-set-debug-files:

-check-env:
 [checkenv] Android SDK Tools Revision 24.4.1

-setup:
     [echo] Project Name: .IoTSemplice
  [gettype] Project Type: Android Library

-set-debug-mode:

-debug-obfuscation-check:

-pre-build:

-build-setup:
[getbuildtools] Using latest Build Tools: 23.0.1
     [echo] Resolving Build Target for .IoTSemplice...
[gettarget] Project Target:   Android 5.1.1
[gettarget] API level:        22
     [echo] ----------
     [echo] Creating output directories if needed...
    [mkdir] Created dir: /project/CordovaLib/res
    [mkdir] Created dir: /project/CordovaLib/libs
    [mkdir] Created dir: /project/CordovaLib/bin
    [mkdir] Created dir: /project/CordovaLib/bin/res
    [mkdir] Created dir: /project/CordovaLib/bin/rsObj
    [mkdir] Created dir: /project/CordovaLib/bin/rsLibs
    [mkdir] Created dir: /project/CordovaLib/gen
    [mkdir] Created dir: /project/CordovaLib/bin/classes
    [mkdir] Created dir: /project/CordovaLib/bin/dexedLibs
     [echo] ----------
     [echo] Resolving Dependencies for .IoTSemplice...
[dependency] Library dependencies:
[dependency] No Libraries
[dependency] 
[dependency] ------------------

-code-gen:
[mergemanifest] Merging AndroidManifest files into one.
[mergemanifest] Manifest merger disabled. Using project manifest only.
     [echo] Handling aidl files...
     [aidl] No AIDL files to compile.
     [echo] ----------
     [echo] Handling RenderScript files...
     [echo] ----------
     [echo] Handling Resources...
     [aapt] Generating resource IDs...
     [echo] ----------
     [echo] Handling BuildConfig class...
[buildconfig] Generating BuildConfig class.

-pre-compile:

-compile:
    [javac] Compiling 36 source files to /project/CordovaLib/bin/classes
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
     [echo] Creating library output jar file...
      [jar] Building jar: /project/CordovaLib/bin/classes.jar

-post-compile:

-obfuscate:

-dex:
     [echo] Library project: do not convert bytecode...

-crunch:
   [crunch] Crunching PNG Files in source dir: /project/CordovaLib/res
   [crunch] To destination dir: /project/CordovaLib/bin/res
   [crunch] Crunched 0 PNG files to update cache

-package-resources:
     [echo] Library project: do not package resources...

-package:
     [echo] Library project: do not package apk...

-post-package:

-do-debug:
     [echo] Library project: do not create apk...
[propertyfile] Creating new property file: /project/CordovaLib/bin/build.prop
[propertyfile] Updating property file: /project/CordovaLib/bin/build.prop
[propertyfile] Updating property file: /project/CordovaLib/bin/build.prop
[propertyfile] Updating property file: /project/CordovaLib/bin/build.prop

-post-build:

debug:

-code-gen:
[mergemanifest] Merging AndroidManifest files into one.
[mergemanifest] Manifest merger disabled. Using project manifest only.
     [echo] Handling aidl files...
     [aidl] No AIDL files to compile.
     [echo] ----------
     [echo] Handling RenderScript files...
     [echo] ----------
     [echo] Handling Resources...
     [aapt] Generating resource IDs...
     [echo] ----------
     [echo] Handling BuildConfig class...
[buildconfig] Generating BuildConfig class.

-pre-compile:

-compile:
    [javac] Compiling 9 source files to /project/bin/classes
    [javac] /project/src/com/megster/cordova/ble/central/Peripheral.java:71: error: cannot find symbol
    [javac]             gatt = device.connectGatt(activity, false, this, BluetoothDevice.TRANSPORT_LE);
    [javac]                                                                             ^
    [javac]   symbol:   variable TRANSPORT_LE
    [javac]   location: class BluetoothDevice
    [javac] Note: /project/src/com/megster/cordova/ble/central/BLECentralPlugin.java uses or overrides a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] 1 error

BUILD FAILED
/android-sdk/tools/ant/build.xml:716: The following error occurred while executing this line:
/android-sdk/tools/ant/build.xml:730: Compile failed; see the compiler error output for details.

Total time: 7 seconds

/project/cordova/node_modules/q/q.js:126
                    throw e;
                    ^
Error code 1 for command: ant with args: debug,-f,/project/build.xml

Really dont know what may I do.
Someone may help me.

Thanks

valterforesto, 2016-05-13:
The workaround that solve the issue in this thread it's :
`<preference name="phonegap-version" value="cli-6.1.0" /> `

Add the above "code" to the "config.xml" file before submit it to PhoneGAP BUILD.

don, 2016-05-27:
@valterforesto thanks for the solution. I updated the readme to reflect this change.

closed 2016-05-27 by don