don/cordova-plugin-ble-central #243: Can't write GATT Characteristic Descriptors to start DFU mode

thomaswb, 2016-05-18:
I'm trying to get a Nordic nRF8x chip into DFU mode, but I'm supposed to set the CCCD Client Characteristic Configuration Descriptor first, to enable DFU. This descriptor is described in the BLE standard:

https://www.bluetooth.com/specifications/assigned-numbers/generic-attribute-profile

However, this requires writing to the BLE standard characteristic 0x2902, which does not appear in the list of known characteristics when connecting to the device using ble-central. Writing to that descriptor just gives an error (tried with a number of possible service UUIDs):

Could not find characteristic with UUID 2902 on service with UUID 180A on peripheral with UUID 482BC19C-2132-69F6-6A19-F181A4791D5F

So how can I write to this characteristic?

thomaswb, 2016-05-18:
Actually, this has become less urgent-- we've changed the BLE device firmware not to require this characteristic to go into DFU mode. But the general principle still applies: accessing characteristics that are in the standard but not discovered by the plugin.

don, 2016-05-19:
0x2902 is a descriptor, rather than a characteristic. Wrtitng descriptors will be added in the future. See #111

closed 2016-05-19 by don

tomasbedrich, 2016-09-09:
Any plans to add this in the near future? I have to implement the same behavior.

don, 2018-06-07:
@Deniall check out this fork https://github.com/fxe-gear/cordova-plugin-ble-central for handling DFU mode

Deniall, 2018-06-07:
Hey @don thanks for the reply. I'm actually using noble (node) to complete the DFU process, and I have it set up correctly now, however it would make the process simpler if I didn't have to write 02 to the CCCD first, then 01 to the characteristic to begin DFU, hence why I was wondering how @thomaswb managed to change it to not require.