don/cordova-plugin-ble-central #244: Wrong callback getting called with startLeScan

hellstad, 2016-05-18:
Hi, first of all thanks for the great work on the plugin. I am working on a BLE device based on Nordic NRF51422. This plugin is working well for iOS and newer Android flagships (HTC 10/Marshmallow, HTC M8/Marshmallow, LG G3/Lollipop). However it is unable to find any devices when scanning on some older Androids (Motorola Droid Mini/KitKat, Samsung Galaxy S3/Lollipop).

Testing on the Samsung Galaxy S3, it seems that the wrong callback is getting triggered. After starting the scan, the `onScanResult()` callback is getting triggered (as seen in the console output):

```
05-18 14:41:14.589 22057-23360/com.ionicframework.sw839277 D/BluetoothAdapter: onScanResult() - Device=ED:F3:43:88:3D:17 RSSI=-50
05-18 14:41:14.974 22057-22070/com.ionicframework.sw839277 D/BluetoothAdapter: onScanResult() - Device=ED:F3:43:88:3D:17 RSSI=-50
05-18 14:41:15.169 22057-22069/com.ionicframework.sw839277 D/BluetoothAdapter: onScanResult() - Device=ED:F3:43:88:3D:17 RSSI=-51
05-18 14:41:15.364 22057-23361/com.ionicframework.sw839277 D/BluetoothAdapter: onScanResult() - Device=ED:F3:43:88:3D:17 RSSI=-51
```

Since the plugin is using the older `startLeScan()` method and handles the discovered devices through the `onLeScan()` callback, no devices get reported during scanning.

Is this a resolvable issue - or am I doing something wrong here? I am using Cordova 6.0, minSdk is 18 and targetSdk is 23 (I've tried 19, 21 also).

don, 2016-05-19:
Filtering scan result by service uuid doesn't work well on older Android devices. 

Can you put together a simple project demonstrating the issue? I'll test it on some of my order hardware and see if there is a fix.

hellstad, 2016-05-20:
Yeah, I've read that. I haven't been filtering the scan result at all. I will pull the specific code out of my project (actually we have 2 projects from 2 different devs with the same behaviour) and make a repo for you to test.

hellstad, 2016-05-25:
Here is a simple project, basically just the BLE scan code extracted out of the project I am working on so might have left some redundant stuff in there.

repo: https://github.com/hellstad/cordova-ble-scan-bug

I'm using Cordova 6.0, and project should work with node 4.x~6.x
Thanks for your time.

don, 2016-05-27:
@hellstad I'm seeing an Ionic splash screen then a blank screen. Same results on my Samsung S3 GT-I9300 with 4.3 and a Nexus 5X with 6.0.1.

ionic --version
1.7.14

cordova -v
6.1.1

node -v
v4.4.4

running with 

     npm install && ionic state restore && npm run build && ionic run android --device

Error (from Nexus 5X)
[image]

$ cordova plugin ls
cordova-plugin-ble-central 1.1.0 "BLE"
cordova-plugin-compat 1.0.0 "Compat"
cordova-plugin-console 1.0.3 "Console"
cordova-plugin-device 1.1.2 "Device"
cordova-plugin-file 4.2.0 "File"
cordova-plugin-file-transfer 1.5.1 "File Transfer"
cordova-plugin-splashscreen 3.2.2 "Splashscreen"
cordova-plugin-statusbar 2.1.3 "StatusBar"
cordova-plugin-transport-security 0.1.2 "App Transport Security"
cordova-plugin-whitelist 1.2.2 "Whitelist"
cordova-plugin-zip 3.1.0 "cordova-plugin-zip"
ionic-plugin-keyboard 2.2.0 "Keyboard"

don, 2016-08-19:
@hellstad I'm assuming you got this figured out but now. If not, update this ticket and I'll reopen the issue.

closed 2016-08-19 by don

hellstad, 2016-08-19:
Sorry @don I meant to update this issue. I "fixed" this not-discovering bug on other older phones by using Crosswalk and splicing in the Nordic Android Compat Library for scanning. The bug was compounded by the older versions of WebView not functioning consistently and the BLE stack just not working on some of my test phones so it was hard to pinpoint. (I still have issues with all my Samsung test devices though, but I've given up on them - probably hardware related)

Thank you for your hard work!

don, 2016-08-19:
@hellstad thanks for the update. Interesting approach for older phones. I need to take a look at https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library

hellstad, 2016-08-19:
I did push some of my work on splicing the Nordic Compat Library onto your plugin, but it was put together quite quickly and it could be done better (I just copied the Nordic files manually into the plugin src dir): https://github.com/hellstad/cordova-plugin-ble-central
It helps with discovery on my older phones but these phones struggle to connect/maintain BLE connection anyway.