don/cordova-plugin-ble-central #246: iOS BLE Disconnect Called

kavilla, 2016-05-18:
Hey, let me start off by saying thanks for making this. 

I am using our own ble devices and I am experiencing an issue with disconnect being called on the iOS version of the app. The android version works well and keeps running for indefinite amount of time so it can't be our devices. I added the cordova plugin add cordova-custom-config but I didn't think that would be the issue because I am leaving the app open on the screen (and I was correct). The code that I am running is pretty much similar to the accelerometer example where I subscribe to notifications.  Then I leave it on the screen and I get all the data. Then exactly a minute and half later (everytime), disconnect gets called and the error message just prints out the peripheral information (like name, services, etc).

Is there something I am missing while setting up for iOS to prevent disconnect being called unless a user explicitly presses a disconnect button?

Thank you!

djlitvak, 2016-05-19:
The disconnect is firing from the peripheral device. ble.connect(app.success(reason), app.error(reason) is the line calling our error function.  The problem does not occur on Android.

We have narrowed it down to see that the issue has to do with connection parameters (https://devzone.nordicsemi.com/documentation/nrf51/5.2.0/html/a00186.html). Our device does not believe the connection is still active and causes a disconnect.  This only occurs on iOS.

We believe this may be a potential problem within either/both our firmware and the plugin. We are investigating now into our firmware.

Don, do you any thoughts about this on the plugin side? Does the plugin respond to connection parameter update requests?

don, 2016-05-19:
The plugin is using mostly ios defaults. If you find recommended setting for iOS, we can look at adding a way to pass them from JavaScript. Hopefully you can find a way on the peripheral side to keep ios happy. 

Can you test with the LightBlue Explorer app and see if that has a disconnecting problem too?

djlitvak, 2016-05-19:
Working on testing it now. It appears the iOS connection interval and our firmware connection interval do not agree. We will try making that change and I will report back. We will also take a stab with the LightBlue Explorer app.

djlitvak, 2016-05-19:
The problem was our connection interval being too fast on the peripheral. iOS requires a minimum of 20ms. We were around 10ms. A quick firmware update and we are fixed!

See - https://developer.apple.com/hardwaredrivers/BluetoothDesignGuidelines.pdf

closed 2016-05-24 by kavilla