don/cordova-plugin-ble-central #248: android 6.0 cannot find any devices

qppq54s, 2016-05-20:
hi
My devices update to android 6.0 this days,and when i startScan there are no devices callback.
It worked fine before
Can you help me about this,thanks

don, 2016-05-20:
Ensure you're using the lastest version of the plugin that has support for Android 6 permissions.

qppq54s, 2016-05-20:
thanks for the reply!
I checked the version and It's 1.1.0.
But when i open the 'Open Bluetooth debug log' and 'Open Bluetooth packet log' options in developer options, it works fine. 
Then i close these options, it works fine and never show the problem i met before;
Could you figure out the reason

qppq54s, 2016-05-20:
thanks for the reply!
I checked the version and It's 1.1.0.
My phone is nexus 5X, android 6.0.1 and i can find my bluetooth devices in my system bluetooth setting,but it never gives me a callback when using the plugin;
I have another device MI4 ,android 6.0.1, the plugin works fine
Could you figure out the reason

closed 2016-05-24 by qppq54s

djlitvak, 2016-06-03:
I was having this problem with my Nexus 6p. We were using 1.0.4. I just updated to 1.1.0 and have now been unable to reproduce - I will continue testing.

The strange thing was that on 1.0.4, I was able to find devices if I toggled my Bluetooth off/on through the phone settings.