don/cordova-plugin-ble-central #249: Android // onClientConnectionState() - status=8 clientIf=5

lauleehong, 2016-05-20:
Hi,

Occasionally during scan or connect, I get the following error on my console, and an alert popup on the app:

D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=xx:xx:xx:xx:xx:xx
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=5

I troubleshot using the iOS version side-by-side, and seems fine on the iOS. I understand this is a backend issue with Android perhaps, as status=8 meant that there is a form of Link Loss between central and peripheral?

don, 2016-05-27:
Which function is calling app.onError? The error callback from ble.connect?

I think BLE899753944 is just the callback identifier generated by Cordova.

lauleehong, 2016-05-27:
I get this intermittent error from both ble.scan and ble.connect

don, 2016-05-27:
I think you're just seeing random disconnects. ble.connect will call the failure function when the bluetooth state changes to something other than STATE_CONNECTED. https://github.com/don/cordova-plugin-ble-central/blob/f86b10794e3764423167655a9f8f364bdc02516f/src/android/Peripheral.java#L204 or if the service discovery fails https://github.com/don/cordova-plugin-ble-central/blob/f86b10794e3764423167655a9f8f364bdc02516f/src/android/Peripheral.java#L186

The failure callback includes the JSON description of the peripheral so you can tell which peripheral disconnected (when you have multiple connections.)

I'm not sure much can be done to stop this. There are some devices (and phones) that like to randomly disconnect. I think you're going to need to handle this in your app by adding a dedicated error handler for ble.connect.

I didn't see anywhere where ble.scan was calling the failure callback for Android.

closed 2016-08-18 by don