don/cordova-plugin-ble-central #250: Load bonded devices at initialization of plugin

t4deon, 2016-05-27:
It would be great if all bonded BLE devices are loaded at initialization of the plugin so you don't always have to scan for BLE devices first before connecting. To my knowledge, a huge benefit of BLE is that a peripheral does not always have to advertise but you can connect to it. So in my eyes, it would be much better to get all bonded devices at initialization time of the plugin.

don, 2016-06-16:
This sounds like a good feature. It's not something I'd add until I need it for a project. If you get it working, submit a pull request.

don, 2016-09-17:
I don't think there's a good cross platform way to do this. Android has a way to [get Bonded devices](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#getBondedDevices()) but as far as I can tell, this isn't possible on iOS. It might be possible to [get a list of devices by UUID](https://developer.apple.com/reference/corebluetooth/cbcentralmanager/1519127-retrieveperipheralswithidentifie?language=objc), but it's not clear to me if these are connected device or bonded devices.

djlitvak, 2016-10-13:
@t4deon did you get something like this working? I would like to be able to make a bond and still interact with our bluetooth unit.

don, 2018-06-11:
Added [ble.bondedDevices](https://github.com/don/cordova-plugin-ble-central#bondeddevices) for Android. There's no direct way to get bonded devices on iOS. Try [ble.peripheralsWithIdentifiers](https://github.com/don/cordova-plugin-ble-central#peripheralswithidentifiers) or [ble. connectedPeripheralsWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices).

Note that on recent versions of the plugin with Android, you connect using a MAC address without scanning.

closed 2018-06-11 by don