don/cordova-plugin-ble-central #251: Write data to an arduino UNO board failed

iszufu, 2016-05-28:
Hi, Don, I got some problems with sending data. Here is the code relevent to send data action:
html
`
            <div id = "detailPage">
                <div>
                    <input type="text" id="inputArea" name="inputArea"/>
                    <button type="button" id="sendButton">Send</button>
                </div>
                <button id="disconnectButton">Disconnect</button>
                <div id = "message">Welcome to Mohism</div>
            </div>
`
js
`
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
        refreshButton.addEventListener('touchstart', this.refreshDeviceList, false);
        disconnectButton.addEventListener('touchstart', this.disconnect, false);
        deviceList.addEventListener('touchstart', this.connect, false);
        sendButton.addEventListener('touchstart', this.sendData, false);
    },
`
and
`
    sendData: function () {
        var dataReceived = inputArea.value;
        //alert("data: " + inputArea.value);
        var data = new Uint8Array(1);
        var deviceId = event.target.dataset.deviceId;
        
        var onSend = function () {
            alert("Data sended!");
        };
        
        data[0] = dataReceived;
        alert(data[0] + typeof(data[0]));
        //alert(deviceId);
        //alert(data.buffer);
        //alert(data[0]);
        ble.write(deviceId, dfrobot.service, dfrobot.characteristic, data.buffer, app.onSend, app.onError);
    },
`
I hope to interact with an arduino UNO board, currently just trying to light the 'holy' light (LED). I have successfully connected to the board, and "rx" & "tx" lights would blink when the sendButton being pressed. Unfortunately, the board LED was NOT lighted. The arduino part code is showing below:
`
char buffer[1];
int val = 0;
void setup() {
  Serial.begin(115200);  //initial the Serial
  pinMode(13, OUTPUT);
}
 
void loop() {
  if (Serial.available())  {
    Serial.readBytes(buffer, 1);
    val = (int)buffer[0];
    if (val == '2') {
      digitalWrite(13, HIGH);
    } else {
      digitalWrite(13, LOW);
    }
    //Serial.write(Serial.read());//send what has been received
    //Serial.println();   //print line feed character
  }
}
`
I guess the problem is in arduino code, which might be caused by the formate of data, but I don't konw how to fix it right and can't be sure that is the real reason. By the way, I can ensure there is no problem with service and characteristic uuid.

Hope my explanation is clear to you. Thanks.

don, 2016-06-16:
I assume you were using a DFRobot Bluno. I got it working with some minor changes. I created an on button and off button to simplify the interaction. I send 0x01 to turn on the LED and 0x02 to turn it off. In the Arduino sketch I read one byte at a time.

Note that you can't subscribe to notifications on the Bluno because DFRobot hides the characteristic descriptor. See http://www.dfrobot.com/forum/viewtopic.php?f=18&t=2035&p=9298#p9213 You can read characteristic data, but notifications won't work until they fix the firmware.

If this doesn't fix the problem, add another comment and I'll re-open the issue.

index.html

            <div id="detailPage">
                <div>
                    <button id="onButton">On</button>                    
                    <button id="offButton">Off</button>                    
                </div>
                <button id="disconnectButton">Disconnect</button>
            </div>  

index.js

    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
        refreshButton.addEventListener('touchstart', this.refreshDeviceList, false);
        onButton.addEventListener('click', this.on, false);
        offButton.addEventListener('click', this.off, false);
        disconnectButton.addEventListener('touchstart', this.disconnect, false);
        deviceList.addEventListener('touchstart', this.connect, false); // assume not scrolling
    },

    // ... 

    on: function(event) {
        app.sendData(1);
    },
    off: function(event) {
        app.sendData(0);
    },
    sendData: function(byte) { 

        var data = new Uint8Array(1); // size of 1
        data[0] = byte;

        var success = function() {
            console.log("success");
        };

        var failure = function() {
            alert("Failed writing data to Bluno");
        };

        ble.write(
            app.peripheral.id,
            bluno.serviceUUID,
            bluno.serialCharacteristic,
            data.buffer,
            success, 
            failure
        );
    }

Arduino

    void setup() {
      Serial.begin(115200);
      pinMode(13, OUTPUT);
    }

    void loop() {

      if (Serial.available()) {

        byte data = Serial.read();

        if (data == 0x01) {
          digitalWrite(13, HIGH);
        } else if (data == 0x00) {
          digitalWrite(13, LOW);
        } else {
          Serial.println(data, HEX);
        }
      }

      delay(200); 
    }

closed 2016-06-16 by don

boralbgra, 2020-04-20:
> I assume you were using a DFRobot Bluno. I got it working with some minor changes. I created an on button and off button to simplify the interaction. I send 0x01 to turn on the LED and 0x02 to turn [...]

Hi Don, I am using your library to recieve and write the data to a Bluno via BLE, all the writting part works correctly, but the reading part doesnt. If the bluno sends something by serial, and I try to read it, it will give me the last values that I sended (wrote) to the Bluno all time, and I dont know why!

Have you discover how to receive data from the Bluno?

Thanks

Borja

don, 2020-04-21:
@boralbgra this issue has more details about why the DFRobot Bluno doesn't work. https://github.com/don/cordova-plugin-ble-central/issues/291#issuecomment-240889307

I wasn't able to convince them to make any changes. Based on those instructions, you can modify  the native code in your project so the plugin will work with the Bluno. Unfortunately I can't put those changes into the main plugin since it will break other devices.