don/cordova-plugin-ble-central #252: Send data via Ymodem over Bluetooth

zabojad, 2016-05-30:
Hi @don !

It might not be the best place to ask this question but I need to implement a little app that can update a PCB firmware with the Ymodem protocol over Bluetooth. Is this something possible with this plugin ? Or with another one of the BT Cordova plugins you've produced ?

Thanks in advance for your inputs...

zabojad, 2016-05-31:
I've found this lib that manages the [YModem / XModem protocol](http://yxit.co.uk/source/index.html#Y-Modem).

Do you think it would be feasible to wrap its output in a Bluetooth stream?

don, 2016-06-14:
If you want to send modem commands over BLE, this plugin should work. If you're sending over Bluetooth Classic on Android, try https://github.com/don/BluetoothSerial.

The Cordova plugins will get you a connection to the hardware, hopefully you can get the Ymodem library to give you commands to send over that connection.

closed 2016-06-16 by don