don/cordova-plugin-ble-central #253: Device Name Appears to Cache on Android

adzy2k6, 2016-06-01:
I have currently been using this plugin on Android and iOS. I have a device where the advertised name changes regularly (followed by a complete reset of the device). On iOS, the name gets updated correctly. Android appears to keep using the old name though. I have developed apps directly on android, and not had this problem, so it seems like the issue may be related to how the plugin is using the android BLE classes.

adzy2k6, 2016-06-01:
I have some further information on this. it appears to be related to the length of the device name. It only seems to use the cache for names longer than 7 or 8 characters.

don, 2016-06-05:
Are you changing the advertised name (0x09) or Device Name (0x2a00) in the  Generic Access Service (0x1800)? 

I've had more problems with iOS caching the name until the next connection is made but the problem can happen with Android too. This plugin relies on the behavior of the iOS and Android APIs, so there's not much I can do.

I don't think changing the device name is going to work well because iOS and Android want to cache this data. If you control the peripheral, perhaps you can send your changing value in the Advertising (or Scan response packet) using the Manufacturer Info (0xFF) or by Broadcasting the characteristic value in the Service Data (0x16).

don, 2016-06-15:
Creating a receiver for [ACTION_NAME_CHANGED](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#ACTION_NAME_CHANGED) might be a solution.

don, 2016-06-16:
I can't get the device name to update, but 15a8c204779357c2e3c3bb6370987ea6ff38b92a updates the scanRecord and you can parse the advertised name (0x09) from the advertising data.

     onDiscoverDevice: function(peripheral) {
        var advertising = util.parseAdvertisingData(peripheral.advertising);
        // https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile
        var completeLocalName = util.bytesToString(advertising["0x09"]);
        // ....
    }

Util would look something like 

    var util = {

      bytesToString: function (buffer) {
        return String.fromCharCode.apply(null, new Uint8Array(buffer));
      },

      stringToBytes: function(string) {
            var array = new Uint8Array(string.length);
            var byteStream = '';
            for (var i = 0, l = string.length; i < l; i++) {
                array[i] = string.charCodeAt(i);
                byteStream += ''+ string.charCodeAt(i);
            }
            return array.buffer;
      },

      // i must be < 256
      asHexString: function(i) {
          var hex;

          hex = i.toString(16);

          // zero padding
          if (hex.length === 1) {
              hex = "0" + hex;
          }

          return "0x" + hex;
      },

      // NOTE: this function should be in a future version of cordova-plugin-ble-central
      parseAdvertisingData: function (buffer) {
          var length, type, data, i = 0, advertisementData = {};
          var bytes = new Uint8Array(buffer);

          while (length !== 0) {

              length = bytes[i] & 0xFF;
              i++;

              // decode type constants from https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile
              type = bytes[i] & 0xFF;
              i++;

              data = bytes.slice(i, i + length - 1).buffer; // length includes type byte, but not length byte
              i += length - 2;  // move to end of data
              i++;

              advertisementData[util.asHexString(type)] = data;
          }

          return advertisementData;
      }

    };

closed 2016-08-18 by don