don/cordova-plugin-ble-central #254: Android Disconnect - close vs disconnect

djlitvak, 2016-06-03:
[Peripheral.java#L79](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L79)

`    
public void disconnect() {
connectCallback = null;
        connected = false;
        if (gatt != null) {
            gatt.close();
            gatt = null;
        }
    }
`

Is there a reason that this only calls gatt.close(); and not gatt.disconnect(); then gatt.close();

don, 2016-06-05:
Older versions of the library used gatt.disconnect() but it caused multiple notifications on reconnect. gatt.close() works better.

Are you seeing problems?

djlitvak, 2016-06-05:
I was seeing problems but it was fixed with an update of the plugin.
Android N stuff.

However, it seems to make sense to me that we should be running a
gatt.disconnect followed by gatt.close. That way the peripheral is
disconnected before the connection is closed.

We are running with a system that is going to do many disconnects and
connects before it runs a close.

My understanding of the java docs from aosp is that disconnect allows for a
faster reconnect.

Thoughts?

Thanks a ton for this plugin and the great support!

closed 2016-06-13 by djlitvak