don/cordova-plugin-ble-central #255: Can't get BLE to run in background?

JonasWestAlro, 2016-06-05:
Hi all, 

I'm trying to get the BLE running in background on IOS. I am using this plugin to keep my running in general: https://github.com/katzer/cordova-plugin-background-mode. Works fine. 

And then I've installed the recommended plugin: https://www.npmjs.com/package/cordova-custom-config, but as soon as I add: 

```
<platform name="ios">
    <config-file parent="UIBackgroundModes" target="*-Info.plist">
        <array>
            <string>bluetooth-central</string>
        </array>
    </config-file>
</platform>
```

to my config.xml, the app stops running in background at all? Any ideas? :-)

don, 2016-06-05:
Maybe an error in config.xml? Can you post link to file or add file  to the issue?

JonasWestAlro, 2016-06-05:
@don check updated issue :)

don, 2016-06-05:
Config file looks OK. Are you seeing any errors in Xcode when you background the app?

JonasWestAlro, 2016-06-05:
Hmm, not really.
When I background the app I get these 3 lines in the output window: 

```
2016-06-05 21:33:00.780 MyApp[296:23564] stopScanTimer
2016-06-05 21:33:04.780 MyApp[296:23564] Scanning..
2016-06-05 21:33:04.781 MyApp[296:23564] scan
```

The "Scanning.." comes from the browser app (I have an $interval going that scans every 10sec).

JonasWestAlro, 2016-06-05:
Just realized that the first line comes from a previous scan. The last two lines comes from the interval which fires one more time after I background the app.

don, 2016-06-05:
I've only used the background mode to receive notification from a connected peripheral. I think there's some additional iOS magic or settings to be able to scan for peripherals in the background.

JonasWestAlro, 2016-06-05:
Oh snap. Of cause there are additional iOS magic :-( .. Thanks anyway.

don, 2016-06-06:
Background scanning is a feature I want to support in the plugin, I just haven't needed it yet. I hope to add it in the future.

don, 2016-06-08:
@JonasWestAlro I'm not sure there's any additional stuff to do discovery in the background.

See "The bluetooth-central Background Execution Mode" from https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html

Maybe discovery discovery takes a lot longer in the background?

don, 2016-08-19:
I've confirmed that scanning, connecting and notifications work in the background. See sample project https://github.com/don/ble-background for more details.

closed 2016-08-19 by don