don/cordova-plugin-ble-central #256: iOS: disconnect should remove the notification callbacks

pragmaware, 2016-06-05:
When disconnect() is called right after stopNotification(), the stopNotification callbacks are not removed.
After the next call to connect() the old stopNotification callback may be triggered and depending on BLE stack timing a new notification request may be actually canceled.

In general, disconnect() should probably remove all the existing callbacks, even if they haven't been called.

djlitvak, 2016-06-06:
On Android, in a past native app I wrote, we ran into a similar problem where if we made too many consecutive reads/writes/notifiactions/etc some would not run and not get returns.  I ended up having to make a queue to handle it. [Stack Overflow Discussion](https://stackoverflow.com/questions/21278993/android-ble-how-to-read-multiple-characteristics)

Disconnect was similar in that we waited to empty the queue before disconnecting so that we were certain all necessary interactions had occurred.

pragmaware, 2016-06-06:
Here the problem is not really about lost interactions. It's more about "stale" state being preserved across sessions. A disconnection is usually someting that truncates all the pending interactions.

I agree that one technically *could* first call stopNotifications(), then wait for the callback and then disconnect(). However the fix in the ios source seems to be easy (just remove the callbacks on disconnect) and it would avoid several headaches to developers.

don, 2016-06-14:
@pragmaware I never stopNotifications. Are you stopping notifications because the peripheral is remembering state?

don, 2018-06-07:
This isn't exactly what you wanted, but callbacks are now cleaned up when exiting https://github.com/don/cordova-plugin-ble-central/commit/45989595062b22af28815e04bbbe80efdda9e621

closed 2018-06-07 by don