don/cordova-plugin-ble-central #257: Android N No Devices (Ver 1.1.0)

djlitvak, 2016-06-06:
Don,

When I am scanning for devices with my Nexus 6p on Android N, I do not find any devices. I am on version 1.1.0 with min API 19.

OnClientRegistered is returning a 133 which is a GATT Error but not much detail about that is available online.

Here is the output from ADB - 

06-06 12:57:12.834  7777  7835 D BluetoothAdapter: startLeScan(): null
06-06 12:57:12.836  7777  7835 D BluetoothAdapter: STATE_ON
06-06 12:57:12.840  3438  3456 D BtGatt.GattService: registerClient() - UUID=c5368e25-c8f4-4dcb-ba6e-525dfb9c47ef
06-06 12:57:12.840  3438  4048 E bt_btif : Register with GATT stack failed.
06-06 12:57:12.841  3438  4048 E bt_btif : Register with GATT stack failed.
06-06 12:57:12.841  3438  3546 D BtGatt.GattService: onClientRegistered() - UUID=c5368e25-c8f4-4dcb-ba6e-525dfb9c47ef, clientIf=0
06-06 12:57:12.842  7777  7790 D BluetoothLeScanner: onClientRegistered() - status=133 clientIf=0
06-06 12:57:12.926  7935  7935 I ServiceManager: Waiting for service AtCmdFwd...
06-06 12:57:13.926  7935  7935 I ServiceManager: Waiting for service AtCmdFwd...
06-06 12:57:14.927  7935  7935 I ServiceManager: Waiting for service AtCmdFwd...
06-06 12:57:15.928  7935  7935 W Atfwd_Sendcmd: AtCmdFwd service not published, waiting... retryCnt : 3
06-06 12:57:17.849  7777  7835 D BluetoothAdapter: stopLeScan()
06-06 12:57:17.853  7777  7835 D BluetoothAdapter: STATE_ON
06-06 12:57:17.854  7777  7835 D BluetoothLeScanner: could not find callback wrapper
06-06 12:57:19.035  4704  5230 D LocationFilter: Forcing stable location. Original location:Location[fused 43.074888,-89.386393 acc=20 et=+2d12h54m47s676ms {Bundle[{wifiScan=[B@801cb0c, noGPSLocation=Location[fused 43.074888,-89.386393 acc=20 et=+2d12h50m2s627ms {Bundle[mParcelledData.dataSize=208]}], locationType=3}]}] because SMD didn't trigger.
06-06 12:57:19.072  4704  2620 I GCoreUlr: Successfully inserted 1 locations

djlitvak, 2016-06-06:
When I toggle BLE off then on, I can then find devices.

However, before I toggle BLE, my settings app can find BLE devices.

djlitvak, 2016-06-06:
It seems this has something to do with this: [SCAN_FAILED_APPLICATION_REGISTRATION_FAILED](https://developer.android.com/reference/android/bluetooth/le/ScanCallback.html#SCAN_FAILED_APPLICATION_REGISTRATION_FAILED)

Seems to maybe be something with a race condition happening - [Race Condition Thread](https://code.google.com/p/android/issues/detail?id=173991)

don, 2016-06-14:
@djlitvak Thanks for the info. I don't have time to look at N for a while. If you find more info post it here or send a pull request.

biy-gxnn, 2016-09-06:
I scan for devices on Android, and the result is undefined. How can I find  devices ?

don, 2016-09-17:
@djlitvak  I've been using the plugin on my Nexus 5x with 7.0 and things are working OK. I'm thinking whatever you were seeing with the release candidate is now fixed. If not, please open a new issue with more details.

closed 2016-09-17 by don

kellycampbell, 2017-12-05:
Hey Don, I think we have been running into this same issue. Android logs have:

```
12-05 11:58:35.629  8101  8101 I BeaconBle: Starting scan on OS scanner. BT state: 12
12-05 11:58:35.630  8101  8101 D BluetoothAdapter: STATE_ON
12-05 11:58:35.636 15445  8512 D BtGatt.GattService: registerClient() - UUID=37f10cd3-c178-4131-910c-6869c52664df
12-05 11:58:35.636 15445 15534 E bt_att  : GATT_Register: can't Register GATT client, MAX client reached!
12-05 11:58:35.636 15445 15534 E bt_btif : Register with GATT stack failed.
12-05 11:58:35.636 15445 15534 E bt_att  : GATT_Register: can't Register GATT client, MAX client reached!
12-05 11:58:35.636 15445 15534 E bt_btif : Register with GATT stack failed.
```

That log is actually from using nRF Toolkit since this affects all BLE apps once it starts. I'm digging into it deeper today-tomorrow. First thing I'm trying to address is being able to know when this error happens and notify the user to cycle bluetooth on/off. I think it needs to use the newer android.bluetooth.le package if available, and then hopefully we'll get one of the ScanCallback SCAN_FAILED errors. Thoughts?