don/cordova-plugin-ble-central #258: Can't find other iOS devices

norenma, 2016-06-07:
Hi,

I am trying to create connections between multiple iOS devices. When I scan for bluetooth peripherals I can't seem to find other iOS devices. 

I am using an iPad air and trying to connect other iPad airs. I can find other devices such as my macbook and apple TV, as well of a list of unnamed devices(I am sitting at an office among others.)

colindaly, 2016-06-12:
I have a similar issue. In discovery mode I have;

- Apple BT Keyboard
- Macbook Air
- Apple Pencil
- Thinkpad BT Mouse
- Apple TV

... and the only thing that is getting reported is the Apple TV. When I power off the ATV, none of the other devices are being logged.

Attached is a screenshot from my MacBookAir to confirm they are in discovery mode.
[image]

don, 2016-06-14:
@norenma The devices you want to discover need to be advertising a Bluetooth LE service. If the devices are not advertising services, you won't be able to discover them. You're best off creating Bluetooth LE services with native code. If you're brave, you can try https://github.com/don/cordova-plugin-ble-peripheral

@colindaly it sounds like you're looking in the iOS Bluetooth settings. That's (mostly) for Bluetooth Classic.

closed 2016-06-14 by don