don/cordova-plugin-ble-central #261: Error when bulding cordova

alyeldegwy, 2016-06-14:
-compile:
    [javac] Compiling 6 source files to /home/aly/www/ble/platforms/android/ant-build/classes
    [javac] warning: [options] source value 1.5 is obsolete and will be removed in a future release
    [javac] warning: [options] target value 1.5 is obsolete and will be removed in a future release
    [javac] warning: [options] To suppress warnings about obsolete options, use -Xlint:-options.
    [javac] /home/aly/www/ble/platforms/android/src/com/megster/cordova/ble/central/Peripheral.java:71: error: cannot find symbol
    [javac]             gatt = device.connectGatt(activity, false, this, BluetoothDevice.TRANSPORT_LE);
    [javac]                                                                             ^
    [javac]   symbol:   variable TRANSPORT_LE
    [javac]   location: class BluetoothDevice
    [javac] 1 error
    [javac] 3 warnings
ERROR running one or more of the platforms: Error: /home/aly/www/ble/platforms/android/cordova/run: Command failed with exit code 1
You may not have the required environment or OS to run this project

don, 2016-06-14:
What version of Cordova are you using?
What version of Android SDK?

alyeldegwy, 2016-06-15:
Cordova 6.2.0
Android SDK  24

alyeldegwy, 2016-06-15:
I've updated the Android sdk  and it is working now 
**Thanks Don**

closed 2016-06-15 by alyeldegwy