don/cordova-plugin-ble-central #262: Adafruit Flora Bluefruit LE

floraquestion, 2016-06-19:
Hello,

I’m pretty new to this stuff so please excuse my lack of knowledge, but I’ve recently been working with an Adafruit Flora and Adafruit Flora Bluefruit LE (https://www.adafruit.com/product/2487) and I’m trying to create a pretty simple PhoneGap app where I’ll essentially just need to use a counter and send the counter data to the app when the app requests it. I took a look at your bluefruit le example just to get the framework of how to do this and it looks like you’re using a hardware SPI setup for the bluefruit object. I’ve been learning how to setup a bluetooth connection from https://cdn-learn.adafruit.com/downloads/pdf/adafruit-flora-bluefruit-le.pdf and I’m pretty sure that I’ll need to use a Hardware Serial setup instead of the hardware SPI setup that’s used in the example. 

I was wondering if I edit the callbackEcho.ino code for the bluefruit le example to use a Hardware Serial configuration should I be able to get the app running or will this not work with that setup?

If not, would you happen to know of a different place where I could find an example of how to connect the Flora Bluefruit LE to a PhoneGap app?

closed 2016-06-20 by floraquestion

don, 2016-06-20:
@floraquestion the [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) should would with the [Adafruit Flora Bluefruit LE](https://www.adafruit.com/product/2487).

bethyd, 2017-06-25:
@floraquestion Were you able to get the code to work with the changes, or did you use @don 's suggestion and try it as is?