don/cordova-plugin-ble-central #264: connectFailure callback not being triggered on iOS devices

kerrynh, 2016-06-21:
Hi Don :-)

I'm having some issues with the failure callback on iOS for the following call:

`ble.connect(device_id, connectSuccess, connectFailure);`

According to the docs, 

> "Failure is called if the connection fails, or later if the peripheral disconnects."

If I switch my peripheral off, the connectFailure callback triggers correctly for both Android and iOS. If however, I turn my Bluetooth on the device off, or switch to Airplane mode (which obviously switches the Bluetooth off), the callback is only triggered correctly on Android, NOT iOS.

I have tested this on a few different iOS devices - my dev iPhone is running iOS 8.3 and the other iPhones and iPads were all running various versions of iOS 9. It's the same across the board.

I'm currently running Cordova 5.4.1 and the plugin version is the latest, 1.1.1. In an attempt to solve the issue, I tried reverting to previous plugin versions (down to about 1.0.1) and Cordova version (to 5.1.1), and combinations of that - I thought this was working a few months back....  Still no luck on triggering the callback on iOS. If it was working, I have no idea how or when it broke! Or if I was just thinking of Android...

I'm not sure if it's relevant, but I'm using IntelXDK 3400 to build my app for Android and iOS...

Do you perhaps have any ideas as to how to solve this? I'm contemplating running a `ble.isConnected` check on a timer (every 5 seconds or so), but I'm hoping you have a more elegant suggestion!

Thanks in advance :-) You've helped so many of us with this creating and maintaining this plugin...

don, 2016-06-22:
I'd expect the disconnect event should fire when Bluetooth is turned off. It sounds like a bug in my iOS code. I need to duplicate and fix it.

Until then, you can [listen for state notification changes](https://github.com/don/cordova-plugin-ble-central#startstatenotifications) to tell if and when the user disables Bluetooth.

kerrynh, 2016-06-23:
Thank you so much Don :-) 

Doh! I don't know how I missed the new state notification functionality when I upgraded from 1.0.4... The hazard of trying to do too many things at once... Will definitely do that :-) The `ble.isConnected` timer check hack does work, but your suggestion is far superior!

kof3r, 2016-07-10:
Hi, is there a chance this will be resolved soon? Want to avoid some hacky code :D

don, 2016-07-28:
@kerrynh @kof3r I have a fix in master, can you test it and see if this works for you?

closed 2016-08-18 by don