don/cordova-plugin-ble-central #265: ble is not defined

ka347, 2016-06-24:
hey i am creating an app using telerik platform when i run the app i am getting ble not defined error.

BioPhoton, 2016-06-27:
did you wrap the function with a device ready?

don, 2016-06-27:
That sounds like the plugin is not installed or you're calling it before deviceready fires like @BioPhoton mentioned

DanielSOhara27, 2016-07-13:
@don Do you want us to wrap it inside this code? 

`document.addEventListener("deviceready", yourCallbackFunction, false);` 

because I did add it and the callBack function (your test of the ble plugin) was never fired

`        ble.isEnabled(
                 function() {
                    console.log("Bluetooth is enabled");
                    },
                 function() {
                     console.log("Bluetooth is *not* enabled");
                     alert("Bluetooth is *not* enabled");
                    }
                );`

I also noticed that in an Ionic Environment the App.js file already has a 

`$ionicPlatform.ready(function() { <code goes in here> }` 

So are we supposed to double wrap your BLE function inside two "deviceready" functions?

We've tried reinstalling the cordova plugin, restoring the ionic project, wrapping the ble inside deviceready functions and in the end we always end up with the error "ble is not defined."

Do you have any advice to get your plugin to work?

don, 2016-07-13:
You'll need to check, $ionicPlatform.ready might get called after device ready. You need to ensure Cordova has started before calling any of the functions. 

It could be an ionic scope problem too. For Ionic 1, you'd want the ble code in a service. See https://github.com/don/ionic-ble for one way to do it.

DanielSOhara27, 2016-07-13:
@don 
I tried following the example "ionic-ble" that you posted and still had the same error.

 `ionic.bundle.js:26794 ReferenceError: ble is not defined
    at Object.init (services.js:78)
    at new <anonymous> (OnBoardingCtrl.js:28)
    at Object.instantiate (ionic.bundle.js:18010)
    at $controller (ionic.bundle.js:23412)
    at self.appendViewElement (ionic.bundle.js:59900)
    at Object.render (ionic.bundle.js:57893)
    at Object.init (ionic.bundle.js:57813)
    at self.render (ionic.bundle.js:59759)
    at self.register (ionic.bundle.js:59717)
    at updateView (ionic.bundle.js:65398)`

I tried calling the ble.isEnabled from inside a controller that is scoped correctly with the correct BLE services. Still no good : (  

would you have other suggestions?

Thank you

don, 2016-07-15:
@DanielSOhara27 you got that error with my example code?  It should run as-is without any errors.

I'm using Ionic 1.7.16, Node 4.4.7 and building on OS X 10.11.5. Make sure you see Ionic install the plugin from npm when you add the platform. The plugin is defined in config.xml so it's automatically added. You can verify it's installed with `ionic plugin ls`

    $ ionic plugin ls
    cordova-plugin-ble-central 1.1.1 "BLE"
    cordova-plugin-compat 1.0.0 "Compat"

I haven't been doing a lot of Ionic lately so I can't recall the details, but I don't think you can call `ble.isEnabled` from a controller unless you inject ble. I'd put stuff like this in the service and call the service from the controller.

don, 2016-07-27:
@DanielSOhara27 are you running the app on a real phone?

This won't work in the browser, in an iOS emulator, or in an Android emulator. Bluetooth needs real hardware in order to work.

DanielSOhara27, 2016-07-27:
Hi @don

 In the end, we were able to get the plugin working successfully on both Android and IOS.  I'm sorry I forgot to update this thread and thank you for your help. In the end, it was a problem with our project's settings where cordova was not being loaded correctly and the app was not being bootstrapped. 

Thanks

don, 2016-07-27:
Thanks for the update.

closed 2016-07-27 by don