don/cordova-plugin-ble-central #266: BLE vs bluetoothSerial write

mareksip, 2016-06-27:
When using bluetoothSerial I am able to execute printers ESC commands from Android device with following code

```
dataAlignCenter = new Uint8Array(4);
dataAlignCenter[0] = 0x1B;
dataAlignCenter[1] = 0x1D;
dataAlignCenter[2] = 0x61;
dataAlignCenter[3] = 0x1;

bluetoothSerial.write(dataAlignCenter.buffer, onSuccess, onFailure);

/* being executed on Android */
```

However, same ArrayBuffer is not executed on printer from iOS with ble-central. Connection is properly established and strings are printed when transferred to ArrayBuffer. 

```
ble.writeWithoutResponse(bleDeviceId, bleServiceId, bleCharId, dataAlignCenter.buffer, null, null);

/* being executed on iOS */
```

The ble.writeWithoutResponse does not execute the ESC command. I wonder if there is any difference between ble and bluetoothSerial buffer writing?

I am working with Star Micronics SM-L200 printer

closed 2016-06-27 by mareksip

bjornoss, 2017-02-22:
Hi @mareksip 
Sorry to commenting on an old issue but I want to ask you a question. 
I have the exact same printer as you. I just got it a few days ago and I find it difficult to start printing using the BLE plugin. 
Do you have any code lines you could share and maybe you have a link to some documentation on the command values to use to get me started?
Maybe just the lines to print "Hello World!" or something.  
I would really appreciate it :-) Thanks

lbngoc, 2020-01-13:
Hi @bjornoss ,
I have faced this issue today.
I use [this lib](https://www.npmjs.com/package/esc-pos-encoder-ionic) to generate the ArrayBuffer for writting. It was working perfect with bluetoothSerial plugin but when I changed to BLE, the ESC printer was not print same result when I use bletoothSerial. Could you please give me some advices ?