don/cordova-plugin-ble-central #270: Invalid UUID

ColeHud, 2016-07-01:
I'm struggling while trying to write to some characteristics on a ble device. I've successfully written to 1 characteristic in the past, but I've been running into this error when I try to write to multiple characteristics in succession. I'm running each write as a callback for the previous ones, so that should be fine. I've attached the console output. One problem that I noticed is that it isn't printing out the characteristics even though I used JSON.stringify on the device object. My repo is here:

https://github.com/lab11/eink

with the important file in js/index.js

What exactly is going wrong?
Thanks! Any help would be appreciated.

[image]

`I/chromium(16493): [INFO:CONSOLE(173)] "CLICKED!", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "onEnable", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "Connection error: [object Object]", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "CONNECTION SUCCESSFUL", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "{
I/chromium(16493):   "id": "C0:98:E5:00:F8:02",
I/chromium(16493):   "advertising": {},
I/chromium(16493):   "rssi": -64,
I/chromium(16493):   "advertisement": {
I/chromium(16493):     "serviceUuids": [
I/chromium(16493):       "FEAA"
I/chromium(16493):     ],
I/chromium(16493):     "serviceData": [
I/chromium(16493):       {
I/chromium(16493):         "uuid": "FEAA",
I/chromium(16493):         "data": {
I/chromium(16493):           "0": 16,
I/chromium(16493):           "1": 186,
I/chromium(16493):           "2": 2,
I/chromium(16493):           "3": 103,
I/chromium(16493):           "4": 111,
I/chromium(16493):           "5": 111,
I/chromium(16493):           "6": 46,
I/chromium(16493):           "7": 103,
I/chromium(16493):           "8": 108,
I/chromium(16493):           "9": 47,
I/chromium(16493):           "10": 112,
I/chromium(16493):           "11": 75,
I/chromium(16493):           "12": 105,
I/chromium(16493):           "13": 76,
I/chromium(16493):           "14": 87,
I/chromium(16493):           "15": 53
I/chromium(16493):         }
I/chromium(16493):       }
I/chromium(16493):     ],
I/chromium(16493):     "flags": 6
I/chromium(16493):   }
I/chromium(16493): }", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "created buffer", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "started write", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "error: Invalid UUID: E528A411FF4F3089D44F7CB505ABA641", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "error: Invalid UUID: E528A411FF4F3089D44F7CB505ABA641", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "error: Invalid UUID: E528A411FF4F3089D44F7CB505ABA641", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "error: Invalid UUID: E528A411FF4F3089D44F7CB505ABA641", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(16493): [INFO:CONSOLE(173)] "error: Invalid UUID: E528A411FF4F3089D44F7CB505ABA641", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)`

don, 2016-07-01:
UUIDs aren't valid without the dashes
     
    // OK
    var textUuid   = "E528A410-FF4F-3089-D44F-7CB505ABA641"; 
    // INVALID
    var xcoordUuid = "E528A411FF4F3089D44F7CB505ABA641";

A [peripheral object](https://github.com/don/cordova-plugin-ble-central#peripheral-data) is passed to the success callback for `ble.connect`. If you log that object `console.log(JSON.stringify(peripheral)` you'll get a list of the characteristics and properties. Make sure your code matches the details from the peripheral.

ColeHud, 2016-07-01:
I'll try this out at work tomorrow (later today I guess it is now). Thanks for the quick response!

ColeHud, 2016-07-01:
I tried out the suggested changes and started getting 'error: 13' on write. I read through a previous issue and switched to useing `ble.writeWithoutResponse()` instead of `ble.write()`. 

Unfortunately, I'm still getting errors. Now it's `characteristic not found` even though that exact characteristic was just printed out a few lines earlier as part of the `peripheral` object you mentioned.

What could be causing this?

[image]

don, 2016-07-01:
The characteristic properties control which functions you can use. If the property doesn't match, you'll get the characteristic not found error. (I should fix that error message.)

I see the error in the screenshot, but I can't see the complete peripheral description. Can you post the complete JSON output here?

ColeHud, 2016-07-01:
[image]
[image]

ColeHud, 2016-07-01:
My coworker informed me that the screenshots aren't up to par, so here's the output in text form.

```json
I/chromium(13073): [INFO:CONSOLE(173)] "CLICKED!", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "onEnable", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "CONNECTION SUCCESSFUL", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "{
I/chromium(13073):   "id": "C0:98:E5:00:F8:02",
I/chromium(13073):   "advertising": {},
I/chromium(13073):   "rssi": -68,
I/chromium(13073):   "services": [
I/chromium(13073):     "1800",
I/chromium(13073):     "1801",
I/chromium(13073):     "e528a44a-ff4f-3089-d44f-7cb505aba641"
I/chromium(13073):   ],
I/chromium(13073):   "characteristics": [
I/chromium(13073):     {
I/chromium(13073):       "service": "1800",
I/chromium(13073):       "characteristic": "2a00",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Read",
I/chromium(13073):         "Write"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "1800",
I/chromium(13073):       "characteristic": "2a01",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Read"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "1800",
I/chromium(13073):       "characteristic": "2a04",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Read"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "characteristic": "e528a411-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Write"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "characteristic": "e528a412-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Write"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "characteristic": "e528a413-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Write"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "characteristic": "e528a410-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Write"
I/chromium(13073):       ]
I/chromium(13073):     },
I/chromium(13073):     {
I/chromium(13073):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "characteristic": "e528a414-ff4f-3089-d44f-7cb505aba641",
I/chromium(13073):       "properties": [
I/chromium(13073):         "Write"
I/chromium(13073):       ]
I/chromium(13073):     }
I/chromium(13073):   ]
I/chromium(13073): }", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "created buffer", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "started write", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "error: Characteristic e528a411-ff4f-3089-d44f-7cb505aba641 not found.", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "error: Characteristic e528a411-ff4f-3089-d44f-7cb505aba641 not found.", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "error: Characteristic e528a411-ff4f-3089-d44f-7cb505aba641 not found.", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "error: Characteristic e528a411-ff4f-3089-d44f-7cb505aba641 not found.", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
I/chromium(13073): [INFO:CONSOLE(173)] "error: Characteristic e528a411-ff4f-3089-d44f-7cb505aba641 not found.", source: https://rawgit.com/lab11/eink/master/software/summon/plugins/cordova-plugin-console/www/console-via-logger.js (173)
```

don, 2016-07-01:
Your `a411` characteristic is write so you should change [your code](https://github.com/lab11/eink/blob/32e1ffae8403ad223b941cc68e93fd837f22d005/software/summon/js/index.js#L48) to use `ble.write`.

    {
      "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
      "characteristic": "e528a411-ff4f-3089-d44f-7cb505aba641",
      "properties": [
        "Write"
      ]
    },

ColeHud, 2016-07-01:
Thanks, I'll try that out

ColeHud, 2016-07-01:
When I run it with just ble.write I get error 13

```

I/chromium(18547): [INFO:CONSOLE(296)] "CLICKED!", source:  (296)
I/chromium(18547): [INFO:CONSOLE(215)] "onEnable", source:  (215)
I/chromium(18547): [INFO:CONSOLE(73)] "CONNECTION SUCCESSFUL", source:  (73)
I/chromium(18547): [INFO:CONSOLE(74)] "{
I/chromium(18547):   "id": "C0:98:E5:00:F8:02",
I/chromium(18547):   "advertising": {},
I/chromium(18547):   "rssi": -56,
I/chromium(18547):   "services": [
I/chromium(18547):     "1800",
I/chromium(18547):     "1801",
I/chromium(18547):     "e528a44a-ff4f-3089-d44f-7cb505aba641"
I/chromium(18547):   ],
I/chromium(18547):   "characteristics": [
I/chromium(18547):     {
I/chromium(18547):       "service": "1800",
I/chromium(18547):       "characteristic": "2a00",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Read",
I/chromium(18547):         "Write"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "1800",
I/chromium(18547):       "characteristic": "2a01",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Read"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "1800",
I/chromium(18547):       "characteristic": "2a04",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Read"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "characteristic": "e528a411-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Write"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "characteristic": "e528a412-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Write"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "characteristic": "e528a413-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Write"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "characteristic": "e528a410-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Write"
I/chromium(18547):       ]
I/chromium(18547):     },
I/chromium(18547):     {
I/chromium(18547):       "service": "e528a44a-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "characteristic": "e528a414-ff4f-3089-d44f-7cb505aba641",
I/chromium(18547):       "properties": [
I/chromium(18547):         "Write"
I/chromium(18547):       ]
I/chromium(18547):     }
I/chromium(18547):   ]
I/chromium(18547): }", source:  (74)
I/chromium(18547): [INFO:CONSOLE(78)] "created buffer", source:  (78)
I/chromium(18547): [INFO:CONSOLE(80)] "started write", source:  (80)
I/chromium(18547): [INFO:CONSOLE(55)] "error: 13", source:  (55)
I/chromium(18547): [INFO:CONSOLE(55)] "error: 13", source:  (55)
I/chromium(18547): [INFO:CONSOLE(55)] "error: 13", source:  (55)
I/chromium(18547): [INFO:CONSOLE(55)] "error: 13", source:  (55)
I/chromium(18547): [INFO:CONSOLE(55)] "error: 13", source:  (55)
```

don, 2016-07-02:
Error 13 is GATT_INVALID_ATTRIBUTE_LENGTH

https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#GATT_INVALID_ATTRIBUTE_LENGTH

Try writing a shorter value. If you have the docs for the peripheral, it might tell you the expected and max characteristic size.

ColeHud, 2016-07-05:
I set up the characteristic on the peripheral to accept up to 256 byte arrays. I'm currently sending it something in the range of ~5 byte arrays, but I still get this error. When I use an app to write to the characteristic, it does just fine. Is this error just for the buffer, or could it be for the ids as well? The buffer shouldn't be causing a problem.

don, 2016-07-05:
That error is from Android, the plugin is just passing the error along to you. The MTU for a characteristic is usually 20 bytes. The peripheral need to support large characteristics, some only support 20 bytes. iOS and Android will chunk data and send blob requests for larger characteristics. iOS and new Android devices are pretty solid up to the max of 500 bytes. With older android versions/devices I've seen them fail with over 100 bytes.

If  the app that works runs on your Android phone, they might be formatting the data different or chunking the data different. Try enabling the HCI debug log in Android settings and see what's happening.

ColeHud, 2016-07-05:
I ended up figuring out that error! It actually turned out to be a pretty stupid error on my part.

ColeHud, 2016-07-05:
but now I'm getting the error "bad base-64"  :(

ColeHud, 2016-07-05:
If I'm writing a uint8 with ble.write(), how should I format that int? Should I convert it to a string and then a buffer? When I put it in by itself with no conversion I get "bad base-64"

don, 2016-07-06:
var number = 42;
    var data = new Uint8Array([number]);
    ble.write(data);

closed 2016-07-13 by don