don/cordova-plugin-ble-central #271: Error: cannot find symbol

alienbrett, 2016-07-01:
When running the command
`cordova build`
with the plugin installed, I get the following error and the build fails:

`C:\pathToProject\phonegap\platforms\android\src\com\megster\
cordova\ble\central\Peripheral.java:71: error: cannot find symbol
            gatt = device.connectGatt(activity, false, this, BluetoothDevice.TRA
NSPORT_LE);
                                                                            ^
  symbol:   variable TRANSPORT_LE
  location: class BluetoothDevice
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\pathToProject\phonegap\platforms\android\src\org\ap
ache\cordova\file\AssetFilesystem.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
1 error
:compileDebugJava FAILED

FAILURE: Build failed with an exception.
`
The implementation builds without the plugin but the app throws errors when calling the BLE functions (not surprising).

don, 2016-07-01:
Is your Android SDK up to date? Older versions of Android don't have support for Bluetooth LE.

alienbrett, 2016-07-01:
I have Android 4.4 (19) up through Android 6.0 (23) installed.

don, 2016-07-01:
This sounds like a duplicate of #261 

Updating the Android SDK was the solution. Unfortunately I don't have details of what exactly @alyeldegwy did to fix it.

alienbrett, 2016-07-01:
I fixed it by updating npm and cordova and then re-adding the android platform in cordova. Then I commented out the  `android:icon="@mipmap/ic_launcher"` tag in AndroidManifest.xml. It built fine after that.

closed 2016-07-01 by alienbrett

kelbongoo, 2016-07-16:
For info, I had this problem as well and an evenings worth of Android/node updates and other tinkering didn't solve it.
The only thing that worked for me was to fix the version at 1.0.6.