don/cordova-plugin-ble-central #273: peripheral.characteristics shows properties as Read/Write when it should be Read/Notify

dkarchmer, 2016-07-10:
After calling
```
ble.connect(deviceId,
        function (peripheral) {
            $log.log(peripheral)
        }, ...
);
```
My `peripheral.characterics[0].properties` show `[Read, Write]`.
But it should be `[Read, Notify]`.
I have confirmed using a MacOS native app and a NodeJs script running NOBLE that it is in fact Read/Notify.

If it helps, I am reading an Arduino 101, where it is running a program that is doing:

```
BLEPeripheral blePeripheral;       // BLE Peripheral Device (the board you're programming)
BLEService theService("19b10000e8f2537e4f6cd104768a1214"); 
BLELongCharacteristictheChar("19b10001e8f2537e4f6cd104768a1214", BLERead | BLENotify);   
```

Is this a bug? Or am I missing something basic?

don, 2016-07-10:
@dkarchmer That doesn't sound right. The plugin gets the data from iOS (or Android). Sounds like there's a bug somewhere.

Can you send me a link to your Arduino 101 code and Cordova code? Github works best, but email a zip is OK too.

dkarchmer, 2016-07-10:
Arduino Code is https://gist.github.com/dkarchmer/c0e806deb6f40018dfa0c5268fa190a0
Let me figure out if I can share the Ionic app, or how to share something

dkarchmer, 2016-07-10:
Added this to the ionic-ble example to show it:
https://github.com/dkarchmer/ionic-ble/commit/476cbf2154b264c99fd80b102646b743f42204e0

```
    <div class="list card">
       <ion-item class="item item-divider">
         Characteristics
       </ion-item>
       <ion-item class="item item-icon-right" 
                 ng-repeat="(key, data) in device.characteristics"
                  href="#/main/ble/{{device.id}}/{{data.service}}/{{data.characteristic}}">
         <p><small style="font-size: 70%;">{{ data.service }}</small></p>
         <p><small style="font-size: 70%;">{{ data.characteristic }}</small></p>
         <p><small style="font-size: 75%;">{{ data.properties }}</small></p>
         <i class="icon ion-chevron-right icon-accessory"></i>
       </ion-item>
     </div>
```

don, 2016-07-13:
@dkarchmer I can't duplicate this. I get the same results from the LightBlue and the plugin. Maybe you were seeing some cached results.

Modified Arduino 101 Code
https://gist.github.com/don/32c49cc60245a962aa9c67597bd7ab06

Cordova App
https://github.com/don/ITP-BluetoothLE/tree/master/phonegap/connect

    $ git clone git@github.com:don/ITP-BluetoothLE.git
    $ cd ITP-BluetoothLE/phonegap/connect/
    $ cordova plugin add cordova-plugin-ble-central
    $ cordova plugin add cordova-plugin-dialogs
    $ cordova platform add ios
    $ open platforms/ios/Connect.xcodeproj/

(deploy to phone from Xcode 7.3)

[image]

[image]

don, 2016-07-13:
Similar results on Android 5.1

[image]

dkarchmer, 2016-07-13:
I will try one more time, but I did confirm I can properly connect to other non-arduino IoT devices so this may be something specific about the one Arduino board I was playing with.
I will check other BLE Apps on my iPhone to confirm it is related to HW and not the library.

dkarchmer, 2016-07-13:
Tried again today (nothing changed), but it is now ok. I am guessing my phone was maybe caching something, as I tried it several times, both from the app I was building, as well as a couple other apps, and all show Read/Write.

Anyway, I apologize for the false alarm, and thank you for looking into this.

closed 2016-07-13 by dkarchmer